robotology/yarp #1747: IJoypadEventDriven deprecated

mebbaid, 2018-06-14:
**Describe the bug**
When clean building yarp, and invoking `IJoypadController` from a walking module the following error apears:
```
In file included from /usr/local/src/robot/yarp/build/src/libYARP_conf/include/yarp/conf/api.h:13:0,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/api.h:13,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/LogStream.h:16,
                 from /usr/local/src/robot/element_capture-point-walking/code/cpp/Joypad_module/src/main.cpp:10:
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/IJoypadController.h:257:33: error: expected unqualified-id before ‘[’ token
     explicit IJoypadEventDriven(YARP_DEPRECATED_MSG("Use IJoypadEventDriven(double)") int rate);
                                 ^
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/IJoypadController.h:257:33: error: expected ‘)’ before ‘[’ token
Joypad_module/CMakeFiles/WalkingJoypadModule.dir/build.make:62: recipe for target 'Joypad_module/CMakeFiles/WalkingJoypadModule.dir/src/main.cpp.o' failed
make[2]: *** [Joypad_module/CMakeFiles/WalkingJoypadModule.dir/src/main.cpp.o] Error 1
```
However if i remove the `YARP_DEPRECATED_MSG` from the corresponding file and build yarp again, and then re-build my walking module, the error doesn't appear.

**To Reproduce**
* You have to have the `element_capture-point-walking` module, precisely the `virtualizer/CVirt` Fork/branch.
* Build the source code and you will get the error.

**Configuration (please complete the following information):**
 - OS: ubuntu
 - yarp version: latest devel 3.0.0
-  compiler: gcc 5.4.0

@aerydna

Nicogene, 2018-06-14:
@drdanz the placement of `YARP_DEPRECATED_MSG` seems give problems.

traversaro, 2018-06-14:
I guess to reproduce it is sufficient to compile any example that includes `yarp/dev/IJoypadController.h` using C++14 .

drdanz, 2018-06-14:
Probably we need yet another macro... something like `YARP_DEPRECATED_CONSTRUCTOR`?

jeljaik, 2018-07-06:
Same problem here for our project. We will just get rid of it manually. It doesn't appear when compiling with clang.

PR #1784 (2018-07-09): Fix deprecated constructor in IJoypadEventDriven
It fixes #1747 

Please review code.
commit f98facaa (2018-07-09): IJoyPadController: fix deprecation message in the constructor for gcc compiling with c++14

It fixes: #1747
closed 2018-07-09 by drdanz