robotology/yarp #1749: Negative VOCAB in Image.h

drdanz, 2018-06-14:
As noticed by @jgvictores (https://github.com/robotology/yarp/commit/185a9c36334860c5f655cf25b3766a7c680cf215#r29116913) there is a negative `VOCAB` in Image.h

https://github.com/robotology/yarp/blob/185a9c36334860c5f655cf25b3766a7c680cf215/src/libYARP_sig/include/yarp/sig/Image.h#L63-L65

@aerydna is this wanted or just a typo?

Nicogene, 2018-06-15:
I checked and it is there since `yarp 2.3.65` :scream: 
But I think it was a typo.
@barbalberto what do you think about?

barbalberto, 2018-06-15:
Negative pixelsize were allowed for "custom weird types", but negative `VOCAB` does not make sense to me.
I agree, probably that's a typo.

jgvictores, 2018-06-15:
Yes, it seems it was introduced way back at 0a193e5... only issue here is that the fix definitely breaks b/w compat.

PR #1761 (2018-06-21): Fix negative vocab
This PR fixes #1749.

It will break probably the ABI compatibility but I didn't find another way to fix it.

Please review code.
closed 2018-06-21 by drdanz