robotology/yarp #1751: auto instead of iterator in YARP

Nicogene, 2018-06-15:
We should all the `iterator` usages with `auto`.

`auto` has the twofold advantage, is shorter than `std::map<std::string, MyClass>::iterator`, and the it prevent from implicit conversion because using the wrong type of the iterator.
`auto` will guarantee that no implicit conversion is involved.

It is actually a minor improvement, and it has the drawback that the code became less readable.

francesco-romano, 2018-06-15:
As already discussed in another issue, I think that the use of `auto` in the iterator is ok (the trade-off pays quite well).

Just be careful of the possibility of
`const auto& it` or `auto it` or `auto& it` or there is also I think the double ampersand `auto &&it`

drdanz, 2019-03-22:
A lot of these were ported using clang-tidy, see 73df6c3f7751f886c030d652b310f3b6986aa0cd
I cannot exclude that there might be more hidden, but we will fix them when we find them

closed 2019-03-22 by drdanz