robotology/yarp #1760: Use enum class instead of enum

Nicogene, 2018-06-19:
Replace `enum` with `enum class` in `Image` and `PointCloud`.
This will cause the enforcing on the type, but it will break the API.

randaz81, 2018-10-01:
Beware of this issue: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=43407 
occurring in OLD gcc compilers (e.g. 5.4) installed on Ubuntu 16.04
Fixed in NEW gcc in Ubuntu 18.04.

drdanz, 2018-10-10:
We already have a `YARP_DISABLE_CLASS_ENUM_API_WARNING` macro to disable the warning when needed, and enum classes are already used in some places, see for example
https://github.com/robotology/yarp/blob/f5e903f58cd4550504d6f6643e0bb3f91ece2ffe/src/libYARP_dev/include/yarp/dev/PidEnums.h#L27-L44