robotology/yarp #1765: MultipleAnalogSensorsRemapper fails to build the remapping of sensor names

nunoguedelha, 2018-06-21:
###  Bug description
We create a configuration of sensors that uses the new Multiple Analog Sensors devices and interfaces, remap them as a new `MultipleAnalogSensorsRemapper` device and connect them to a **Yarp** port through a `MultipleAnalogSensorsServer` wrapper. The remapper fails to build the mapping between the sensors indexes in the physical boards and the desired ordered list at the remapper interface level.

### To Reproduce
We can reproduce the issue by configuring a set of inertial sensors on the iCub robot, as per the following steps : 
- configure a `embObjIMU` type device to read the measurements of an IMU from a **strain2** board (ex: file `<robots-configuration-root>/iCubGenova04/hardware/inertials/left_leg-eb7-IMU.xml`)
- in that config file, name the 4 IMU internal sensors : "acc1", "gyr1", "eul1" and "mag1"
- connect a multiple-analog-sensors-remapper to the sensor device. We can define as re-ordered list a single sensor e.g. "gyr1". (ex: file `<robots-configuration-root>/iCubGenova04/hardware/inertials/left_leg-inertials_remapper.xml`)
- connect a wrapper to the remapper (ex: file `<robots-configuration-root>/iCubGenova04/hardware/inertials/left_leg-inertials_wrapper.xml`)
- include these files paths in the robot interface ini file
- run the robot interface.

### Expected behavior
The robot interface should fully start-up. It fails as the remapper fails and displays the error : 
```
"MultipleAnalogSensorsRemapper: impossible to find sensor name gyr1"
```

### Configuration (please complete the following information)
 - OS: Ubuntu (machine green iCub, iCub30)
 - yarp version: 3.0.XX
-  compiler:
I will provide more information on the configuration by tomorrow.

traversaro, 2018-06-21:
From our f2f conversation, I understood you isolated the problem to be the a problem in the implementation of the `getThreeAxisGyroscopeName` in `embObjIMU` .

nunoguedelha, 2018-06-21:
### Analysis and root cause

The remapper processing fails in the method `MultipleAnalogSensorsRemapper::genericAttachAll(...)` at https://github.com/robotology/yarp/blob/6f63933103a7b84244c2050eb6a96ffb395bd3e8/src/devices/multipleanalogsensorsremapper/MultipleAnalogSensorsRemapper.cpp#L205

In this section, we are building the mapping between each sensor name and a pair **[subdevice index, sensor index]**, the list of names being defined in the remapper config file.
For each `sensorType`, the mapping is filled into `m_indicesMap[sensorType]` as a list of `SensorInSubDevice` objects (subdevice/sensor indexes).
The source is the more generic mapping `sensorLocationMap` built in a previous loop https://github.com/robotology/yarp/blob/6f63933103a7b84244c2050eb6a96ffb395bd3e8/src/devices/multipleanalogsensorsremapper/MultipleAnalogSensorsRemapper.cpp#L165 where all sensor names and indexes are collected. That's where `genericGetName(...)` returns empty names https://github.com/robotology/yarp/blob/6f63933103a7b84244c2050eb6a96ffb395bd3e8/src/devices/multipleanalogsensorsremapper/MultipleAnalogSensorsRemapper.cpp#L176, because it tries to get from `m_indicesMap` the index of the subdevice pointing  to the right method `getThreeAxisGyroscopeName`. But, as far as I understood, at this point, `m_indicesMap` is still empty.

There are some assumptions in the analysis I didn't have the opportunity to verify, I'll check tomorrow, along with a fix by replacing the call to `genericGetName(...)` by :
```
MAS_CALL_MEMBER_FN(subDeviceVec[p], getNameMethodPtr)(s,name)
```

nunoguedelha, 2018-06-21:
> From our f2f conversation, I understood you isolated the problem to be the a problem in the implementation of the getThreeAxisGyroscopeName in embObjIMU .

@traversaro , yes indeed, that's what I thought, but it is working properly if I just use a sensor Device (`embObjIMU`) and a wrapper, (so without remapping the sensors), I can get the metadata through the `rpc` interface. (I did that just before leaving).

traversaro, 2018-06-21:
I thought I actually tested the remapper, but instead I found just this empty body: https://github.com/robotology/yarp/blob/master/tests/libYARP_dev/MultipleAnalogSensorsInterfacesTest.cpp#L31 . Given that there are no test on it, indeed it could be a problem of the remapper itself, sorry for trouble.

nunoguedelha, 2018-06-21:
No trouble. I'm really glad you invested your time on this interface. I don't feel like parsing the ports in Matlab anymore :)

nunoguedelha, 2018-07-08:
Fixed. Closing the issue.

closed 2018-07-08 by nunoguedelha