robotology/yarp #1770: Java Bindings don't compile

Nicogene, 2018-06-25:
**Describe the bug**
The java bindings doesn't compile since #1756 
**To Reproduce**
Enable `CREATE_JAVA` and compiles

** Output **

```bash
/home/ngenesio/yarp/build/generated_src/java/yarpJAVA_wrap.cxx:43754:17: error: use of undeclared identifier 'ThreeAxisGyroscope_EXTENDED_INTERFACE_get'
  result = (int)ThreeAxisGyroscope_EXTENDED_INTERFACE_get();
                ^
/home/ngenesio/yarp/build/generated_src/java/yarpJAVA_wrap.cxx:43942:17: error: use of undeclared identifier 'ThreeAxisLinearAccelerometer_EXTENDED_INTERFACE_get'
  result = (int)ThreeAxisLinearAccelerometer_EXTENDED_INTERFACE_get();
                ^
/home/ngenesio/yarp/build/generated_src/java/yarpJAVA_wrap.cxx:44130:17: error: use of undeclared identifier 'ThreeAxisMagnetometer_EXTENDED_INTERFACE_get'
  result = (int)ThreeAxisMagnetometer_EXTENDED_INTERFACE_get();
                ^
/home/ngenesio/yarp/build/generated_src/java/yarpJAVA_wrap.cxx:44318:17: error: use of undeclared identifier 'OrientationSensor_EXTENDED_INTERFACE_get'
  result = (int)OrientationSensor_EXTENDED_INTERFACE_get();
                ^
/home/ngenesio/yarp/build/generated_src/java/yarpJAVA_wrap.cxx:44535:17: error: use of undeclared identifier 'TemperatureSensor_EXTENDED_INTERFACE_get'
  result = (int)TemperatureSensor_EXTENDED_INTERFACE_get();
                ^
/home/ngenesio/yarp/build/generated_src/java/yarpJAVA_wrap.cxx:44723:17: error: use of undeclared identifier 'SixAxisForceTorqueSensor_EXTENDED_INTERFACE_get'
  result = (int)SixAxisForceTorqueSensor_EXTENDED_INTERFACE_get();
                ^
/home/ngenesio/yarp/build/generated_src/java/yarpJAVA_wrap.cxx:44905:17: error: use of undeclared identifier 'ContactLoadCellArray_EXTENDED_INTERFACE_get'
  result = (int)ContactLoadCellArray_EXTENDED_INTERFACE_get();
                ^
/home/ngenesio/yarp/build/generated_src/java/yarpJAVA_wrap.cxx:45070:17: error: use of undeclared identifier 'EncoderArray_EXTENDED_INTERFACE_get'
  result = (int)EncoderArray_EXTENDED_INTERFACE_get();
                ^
/home/ngenesio/yarp/build/generated_src/java/yarpJAVA_wrap.cxx:45235:17: error: use of undeclared identifier 'SkinPatch_EXTENDED_INTERFACE_get'
  result = (int)SkinPatch_EXTENDED_INTERFACE_get();

```

**Configuration (please complete the following information):**
 - OS: Ubuntu 16.04
 - yarp version: latest devel
-  compiler: clang 3.8

Nicogene, 2018-06-25:
Probably I'have the build messy because I just noticed that the binding of JAVA are compiled on travis and there they work. I will check tomorrow and eventually close it!

Nicogene, 2018-06-26:
I can reproduce it also on a clean build, could be related to the version of swig?
On my machine I have `3.0.8-0ubuntu3`

nunoguedelha, 2018-07-02:
Hi @Nicogene , ...
#### regarding the SWIG version
as we generate bindings for Matlab too, we are actually using a SWIG fork that Silvio and I maintain: https://github.com/robotology-dependencies/swig. The default branch is `matlab` and the latest commit displays the Swig version **3.0.11**. I strongly suggest that you get aligned with our SWIG branch.

#### regarding the BUG
The generated symbols you are seeing are clearly incomplete. You should get `IThreeAxisLinearAccelerometers_`**getNrOf**`ThreeAxisLinearAccelerometers` or `IThreeAxisLinearAccelerometers_getThreeAxisLinearAccelerometer`**Status**, etc, and not just `ThreeAxisLinearAccelerometer_EXTENDED_INTERFACE_`**get**. You might get the right symbols with the updated SWIG.
Let me know how it goes.

Nicogene, 2018-07-02:
Hi @nunoguedelha !

I updated to Bionic Beaver and now I have swig 3.0.12 and the problem persists.

I don't know what could be the problem in my case :thinking:

nunoguedelha, 2018-07-03:
It looks like the preprocessor of SWIG is doing a wrong interpretation of the defines in the interface file `.../robotology-superbuild/robotology/YARP/bindings/macrosForMultipleAnalogSensors.i`.
We actually pulled an old fix on an issue with `%define` and `#define` handling by SWIG: https://github.com/robotology-dependencies/swig/pull/1.

You should really try with the latest commit on the `matlab` branch in the repo in robotology-dependencies: https://github.com/robotology-dependencies/swig.

nunoguedelha, 2018-07-04:
@Nicogene , any news?

Nicogene, 2018-07-11:
Also @drdanz my same issue and he has swig `3.0.12` too.
On travis it works because on travis there ins swig `3.0.2`
So if I understood correctly these changes (robotology-dependencies/swig#1) are necessary to compile the java bindings with newer version of swig than `3.0.2`.
So this is the situation now

|swig version||
|---------------|-|
|`3.0.2`| :white_check_mark:|
|`3.0.8`|:x:|
|`3.0.11`|:x:|
|`3.0.11`(matlab branch)| :white_check_mark:|
|`3.0.12`|:x:|

We have to fix it, we can't assume that the users use this [custom version of swig](https://github.com/robotology-dependencies/swig).

traversaro, 2018-07-11:
I am surprised that 3.0.12 is not working, as the related fix https://github.com/swig/swig/pull/781 should be part of the released version.

drdanz, 2018-07-11:
Tested clean builds with all "official" swig versions from 3.0.0 to 3.0.12:

| swig version             |                    |
|--------------------------|--------------------|
| `3.0.0`                  | :white_check_mark: |
| `3.0.1`                  | :white_check_mark: |
| `3.0.2`                  | :white_check_mark: |
| `3.0.3`                  | :x:                |
| `3.0.4`                  | :x:                |
| `3.0.5`                  | :x:                |
| `3.0.6`                  | :x:                |
| `3.0.7`                  | :x:                |
| `3.0.8`                  | :x:                |
| `3.0.9`                  | :x:                |
| `3.0.10`                 | :x:                |
| `3.0.11`                 | :x:                |
| `3.0.11` (matlab branch) | :white_check_mark: |
| `3.0.12`                 | :x:                |

On travis we have `3.0.2`
The matlab branch is very suspicious, just to be sure, on what system are you building the bindings with the matlab branch, and are you sure that it is not using the **system** share files (i.e. `/usr/share/swig3.0`) instead of the custom build ones?

traversaro, 2018-07-11:
The matlab branch contains explicitly the fix for this issue (https://github.com/swig/swig/pull/781), that was backported in https://github.com/robotology-dependencies/swig/pull/1 .

drdanz, 2018-07-11:
This PR is available both in `3.0.11` and `3.0.12` but none of them seem to work

drdanz, 2018-07-25:
Since we are releasing on YARP 3.1.0 next Monday, we need to fix this ASAP.
@traversaro, @nunoguedelha what do you suggest?

Can we insert the broken part in a `#if !defined(SWIGJAVA)` block?

traversaro, 2018-07-25:
> Can we insert the broken part in a #if !defined(SWIGJAVA) block? 

I guess yes.

nunoguedelha, 2018-07-25:
> Can we insert the broken part in a #if !defined(SWIGJAVA) block?

Seems fine to me too.

drdanz, 2018-07-26:
#1817 disabled the interfaces for java, I renamed the bug, but I think we shouldn't close this until we actually fix the issue, or swig is fixed upstream...

Nicogene, 2018-07-26:
I agree !

Tobias-Fischer, 2018-08-15:
The same happens in C#:
```
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx: In function ‘int CSharp_ThreeAxisGyroscope_EXTENDED_INTERFACE_get()’:
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx:35585:59: error: ‘ThreeAxisGyroscope_EXTENDED_INTERFACE_get’ was not declared in this scope
   result = (int)ThreeAxisGyroscope_EXTENDED_INTERFACE_get();
                                                           ^
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx: In function ‘int CSharp_ThreeAxisLinearAccelerometer_EXTENDED_INTERFACE_get()’:
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx:35743:69: error: ‘ThreeAxisLinearAccelerometer_EXTENDED_INTERFACE_get’ was not declared in this scope
   result = (int)ThreeAxisLinearAccelerometer_EXTENDED_INTERFACE_get();
                                                                     ^
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx: In function ‘int CSharp_ThreeAxisMagnetometer_EXTENDED_INTERFACE_get()’:
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx:35901:62: error: ‘ThreeAxisMagnetometer_EXTENDED_INTERFACE_get’ was not declared in this scope
   result = (int)ThreeAxisMagnetometer_EXTENDED_INTERFACE_get();
                                                              ^
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx: In function ‘int CSharp_OrientationSensor_EXTENDED_INTERFACE_get()’:
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx:36059:58: error: ‘OrientationSensor_EXTENDED_INTERFACE_get’ was not declared in this scope
   result = (int)OrientationSensor_EXTENDED_INTERFACE_get();
                                                          ^
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx: In function ‘int CSharp_TemperatureSensor_EXTENDED_INTERFACE_get()’:
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx:36243:58: error: ‘TemperatureSensor_EXTENDED_INTERFACE_get’ was not declared in this scope
   result = (int)TemperatureSensor_EXTENDED_INTERFACE_get();
                                                          ^
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx: In function ‘int CSharp_SixAxisForceTorqueSensor_EXTENDED_INTERFACE_get()’:
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx:36401:65: error: ‘SixAxisForceTorqueSensor_EXTENDED_INTERFACE_get’ was not declared in this scope
   result = (int)SixAxisForceTorqueSensor_EXTENDED_INTERFACE_get();
                                                                 ^
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx: In function ‘int CSharp_ContactLoadCellArray_EXTENDED_INTERFACE_get()’:
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx:36553:61: error: ‘ContactLoadCellArray_EXTENDED_INTERFACE_get’ was not declared in this scope
   result = (int)ContactLoadCellArray_EXTENDED_INTERFACE_get();
                                                             ^
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx: In function ‘int CSharp_EncoderArray_EXTENDED_INTERFACE_get()’:
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx:36691:53: error: ‘EncoderArray_EXTENDED_INTERFACE_get’ was not declared in this scope
   result = (int)EncoderArray_EXTENDED_INTERFACE_get();
                                                     ^
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx: In function ‘int CSharp_SkinPatch_EXTENDED_INTERFACE_get()’:
/home/tobias/robot/yarp/build2/bindings/csharp/yarpCSHARP_wrap.cxx:36829:50: error: ‘SkinPatch_EXTENDED_INTERFACE_get’ was not declared in this scope
   result = (int)SkinPatch_EXTENDED_INTERFACE_get();
```
I created a patch in https://github.com/robotology/yarp/pull/1830

closed 2025-01-21 by randaz81