robotology/yarp #1771: Vocab error in YarpPluginFixture

diegoferigo, 2018-06-26:
**Describe the bug**
Building yarp fails with the following error:

```
[ 99%] Building CXX object src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/YarpPluginFixture.cpp.o
In file included from /projects/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/SharedLibraryFactory.h:13:0,
                 from /projects/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/SharedLibraryClassFactory.h:12,
                 from /projects/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/SharedLibraryClass.h:12,
                 from /projects/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/YarpPluginSettings.h:14,
                 from /projects/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/YarpPlugin.h:13,
                 from /projects/robotology-superbuild/robotology/YARP/src/rtf-plugins/fixture-managers/yarpplugin/YarpPluginFixture.cpp:15:
/projects/robotology-superbuild/robotology/YARP/src/rtf-plugins/fixture-managers/yarpplugin/YarpPluginFixture.cpp: In function 'int rtf_fixture_factory(void*, int)':
/projects/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/Vocab.h:25:24: error: expected unqualified-id before '(' token
 #define VOCAB(a,b,c,d) ((((int)(d))<<24)+(((int)(c))<<16)+(((int)(b))<<8)+((int)(a)))
                        ^
/projects/robotology-superbuild/build/install/include/rtf/dll/SharedLibraryClassApi.h:84:37: note: in expansion of macro 'VOCAB'
         sapi->startCheck = shlibpp::VOCAB('S','H','P','P'); \
                                     ^~~~~
/projects/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/Vocab.h:25:24: error: expected unqualified-id before '(' token
 #define VOCAB(a,b,c,d) ((((int)(d))<<24)+(((int)(c))<<16)+(((int)(b))<<8)+((int)(a)))
                        ^
/projects/robotology-superbuild/build/install/include/rtf/dll/SharedLibraryClassApi.h:94:35: note: in expansion of macro 'VOCAB'
         sapi->endCheck = shlibpp::VOCAB('P', 'L', 'U', 'G'); \
                                   ^~~~~
src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/build.make:62: recipe for target 'src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/YarpPluginFixture.cpp.o' failed
make[5]: *** [src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/YarpPluginFixture.cpp.o] Error 1
CMakeFiles/Makefile2:5678: recipe for target 'src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/all' failed
make[4]: *** [src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
CMakeFiles/YARP.dir/build.make:108: recipe for target 'robotology/YARP/CMakeFiles/YCMStamp/YARP-build' failed
CMakeFiles/Makefile2:2832: recipe for target 'CMakeFiles/YARP.dir/all' failed
Makefile:94: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[2]: *** [robotology/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/YARP.dir/all] Error 2
make: *** [all] Error 2
```

**Configuration (please complete the following information):**
 - OS: ubuntu:18.04, 16.04
 - yarp version: latest devel @ 2541f1af03273c496eb0ffa9e194067dbe45ef4f
-  compiler: GCC 7.3 and GCC 5.4

**Additional context**

Refer to [this Travis build](https://travis-ci.com/diegoferigo/docker_images/builds/77301520). It's strange because currently the tests on the same commit of this repository are passing.

Nicogene, 2018-06-26:
#1764 fixes this issue, for making compile it again before that the PR has been merged use robot-testing on master branch, the devel contains only the commit that (for now) breaks the rtf addons.
Yet another reason to have integration tests :sweat_smile:

diegoferigo, 2018-06-26:
Thank Nico, good to know! I didn't find any issue about that. Didn't check PRs.

Nicogene, 2018-06-27:
Fixed by #1764 

cc @diegoferigo @pattacini @traversaro

closed 2018-06-27 by Nicogene