robotology/yarp #1772: If YARP_COMPILE_RTF_ADDONS is ON, YARP devel branch is not compiling with RTF devel branch

traversaro, 2018-06-26:
**Describe the bug**
If YARP_COMPILE_RTF_ADDONS is ON, YARP devel branch is not compiling with RTF devel branch .

**To Reproduce**
Download, compile and install RTF in devel branch, then download, compile and INSTALL 

**Expected behavior**
The build is successful.

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04 
 - yarp version: 2541f1af03273c496eb0ffa9e194067dbe45ef4f 
-  compiler: GCC 7.2 

**Detailed error**
~~~
[ 99%] Building CXX object src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/YarpPluginFixture.cpp.o
In file included from /home/straversaro/src/robotology-superbuild-devel/robotology/YARP/src/libYARP_OS/include/yarp/os/SharedLibraryFactory.h:13:0,
                 from /home/straversaro/src/robotology-superbuild-devel/robotology/YARP/src/libYARP_OS/include/yarp/os/SharedLibraryClassFactory.h:12,
                 from /home/straversaro/src/robotology-superbuild-devel/robotology/YARP/src/libYARP_OS/include/yarp/os/SharedLibraryClass.h:12,
                 from /home/straversaro/src/robotology-superbuild-devel/robotology/YARP/src/libYARP_OS/include/yarp/os/YarpPluginSettings.h:14,
                 from /home/straversaro/src/robotology-superbuild-devel/robotology/YARP/src/libYARP_OS/include/yarp/os/YarpPlugin.h:13,
                 from /home/straversaro/src/robotology-superbuild-devel/robotology/YARP/src/rtf-plugins/fixture-managers/yarpplugin/YarpPluginFixture.cpp:15:
/home/straversaro/src/robotology-superbuild-devel/robotology/YARP/src/rtf-plugins/fixture-managers/yarpplugin/YarpPluginFixture.cpp: In function ‘int rtf_fixture_factory(void*, int)’:
/home/straversaro/src/robotology-superbuild-devel/robotology/YARP/src/libYARP_OS/include/yarp/os/Vocab.h:25:24: error: expected unqualified-id before ‘(’ token
 #define VOCAB(a,b,c,d) ((((int)(d))<<24)+(((int)(c))<<16)+(((int)(b))<<8)+((int)(a)))
                        ^
/home/straversaro/src/robotology-superbuild-devel/build/install/include/rtf/dll/SharedLibraryClassApi.h:84:37: note: in expansion of macro ‘VOCAB’
         sapi->startCheck = shlibpp::VOCAB('S','H','P','P'); \
                                     ^~~~~
/home/straversaro/src/robotology-superbuild-devel/robotology/YARP/src/libYARP_OS/include/yarp/os/Vocab.h:25:24: error: expected unqualified-id before ‘(’ token
 #define VOCAB(a,b,c,d) ((((int)(d))<<24)+(((int)(c))<<16)+(((int)(b))<<8)+((int)(a)))
                        ^
/home/straversaro/src/robotology-superbuild-devel/build/install/include/rtf/dll/SharedLibraryClassApi.h:94:35: note: in expansion of macro ‘VOCAB’
         sapi->endCheck = shlibpp::VOCAB('P', 'L', 'U', 'G'); \
                                   ^~~~~
src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/build.make:62: recipe for target 'src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/YarpPluginFixture.cpp.o' failed
make[5]: *** [src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/YarpPluginFixture.cpp.o] Error 1
CMakeFiles/Makefile2:5803: recipe for target 'src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/all' failed
make[4]: *** [src/rtf-plugins/fixture-managers/yarpplugin/CMakeFiles/rtf_fixturemanager_yarpplugin.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/YARP.dir/build.make:108: recipe for target 'robotology/YARP/CMakeFiles/YCMStamp/YARP-build' failed
make[2]: *** [robotology/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
CMakeFiles/Makefile2:2388: recipe for target 'CMakeFiles/YARP.dir/all' failed
make[1]: *** [CMakeFiles/YARP.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
~~~

cc @Nicogene

traversaro, 2018-06-26:
Duplicate of https://github.com/robotology/yarp/issues/1772 , closing.

closed 2018-06-26 by traversaro

nunoguedelha, 2018-06-30:
> Duplicate of #1772 , closing.

ciao @traversaro FYI I guess there is a typo here, this link points to this same issue, not the duplicated one...