robotology/yarp #1777: If YARP is configured with YCM YCM modules are not present in the YARP_MODULE_PATH

traversaro, 2018-07-01:
**Describe the bug**
If YARP is configured without YCM, CMake modules like `FindACE.cmake` are installed and are present in the YARP_MODULE_PATH variable (see https://github.com/robotology/yarp/blob/devel/cmake/template/YARPConfig.cmake.in#L119 and https://github.com/robotology/yarp/blob/devel/cmake/template/YARPConfig.cmake.in#L264).
If instead YARP is configured with YCM, this files are installed by YCM and so they are not available in the YARP_MODULE_PATH . 

**To Reproduce**
Configure YARP `devel` using YCM, with ICUB `devel`, and enable the option `ENABLE_icubmod_embObjMotionControl` (this is just a random device that uses ACE). 

**Expected behavior**
That the behaviour of YARP configured with YCM and YARP configured without YCM is consistent, for example my adding `YCM_MODULE_PATH` to `YARP_MODULE_PATH` in the `YARPConfig.cmake` file.

traversaro, 2018-07-01:
cc @claudiofantacci

claudiofantacci, 2018-07-01:
I just want to report that you can enable any option, even the basic one (`cartesian` and `gaze` controllers) and you will still get the error on `find_package(ACE)`. This is because `ACE` is searched anyway, see [iCubFindDependencies.cmake:52](https://github.com/robotology/icub-main/blob/4f5881acafab693f763dd277f87a8996fdea01a3/conf/iCubFindDependencies.cmake#L52), even if you don't really need it.

PeterBowman, 2018-07-01:
Consistency aside, I believe YARP users are now encouraged to load YCM prior to calling `find_package(ACE)` (or any other supported module). The `YARP_MODULE_PATH` variable is deprecated since YARP 3.0.0 ([ref](https://github.com/robotology/yarp/blob/758d23c4ae8c1affdffdfc0e04830944774a38f6/cmake/template/YARPConfig.cmake.in#L260-L267)). If your devices used to rely on it to enhance the CMake module lookup path, you probably should add a `find_package(YCM)` line in your code.

Nicogene, 2018-07-10:
Fixed by 093d719a4e6cc1d9b4646be86f8b7d59067a42b7

closed 2018-07-10 by Nicogene