robotology/yarp #1802: Serializing maps with thrift idl does not work properly

diegoferigo, 2018-07-13:
### Describe the bug

Let's start with the following thrift:

```thrift

struct SmallContainer {
    1: string name
}

struct BigContainer {
    1: SmallContainer c1,
    2: SmallContainer c2
}

struct test {
    1: map <string, BigContainer> myMap
}
```

The autogenerated class has problems in serializing the `map`, obtaining a serialization that screws up parentheses.

With the help of @drdanz we figured out the problem is due to the nesting of the value of the map (`BigContainer` is a struct containing two structs).

### Possible solution

Digging in the code, here is what I found. Maps are handled by:

https://github.com/robotology/yarp/blob/716b87f9975b85bf459b325e956fb86a9b38ce7a/src/idls/thrift/src/t_yarp_generator.cc#L2565-L2577

And the relevant part of the `generate_serialize_field` is line 2445 of

https://github.com/robotology/yarp/blob/716b87f9975b85bf459b325e956fb86a9b38ce7a/src/idls/thrift/src/t_yarp_generator.cc#L2429-L2451

In this case `generate_serialize_struct` accepts a `force_nesting` flag which is not set for maps. A possible solution would be transforming 

https://github.com/robotology/yarp/blob/716b87f9975b85bf459b325e956fb86a9b38ce7a/src/idls/thrift/src/t_yarp_generator.cc#L2574

to

```cpp
generate_serialize_field(out, &vfield, "", "", true);
```

This seems working also with non-nested structs, but a more thorough testing would be needed. I'm not an expert on this, and I don't know if this would affect other use cases.

### Configuration
 - OS: Ubuntu 18.04
 - yarp version: latest devel @ 716b87f9975b85bf459b325e956fb86a9b38ce7a
-  compiler: clang 5

drdanz, 2018-07-13:
Thanks for the very detailed analysis.
I'm not sure if forcing the nested is always, the right thing to do, when second is not a struct, perhaps it should be serialized normally... We can probably check it whith one of these methods:

https://github.com/robotology/yarp/blob/716b87f9975b85bf459b325e956fb86a9b38ce7a/extern/thrift/thrift/compiler/cpp/src/thrift/parse/t_type.h#L47-L60

Have you checked if by applying your fix, the deserialization part is correct?

Nicogene, 2018-07-16:
The thrift of yarp as another problem related to the pointclouds.
When it is sent as bottle (after the call of `toBottle`), on the receiver side it comes as a Bottle of 1 element that it is a list containing the N points.
Without thrift the bottle received contains N elements as expected(it is tested in yarp unit tests)
@fbottarel has better insight in this issue.
Probably we should open another issue, but I think it is another problem in the de-serialization

drdanz, 2018-07-16:
> I'm not sure if forcing the nested is always, the right thing to do, when second is not a struct, perhaps it should be serialized normally

Actually, the write nested might be already doing this, we should check

diegoferigo, 2018-07-16:
@drdanz I dug a bit more on this problem which is blocking for me at the current moment. Lists and maps are fairly similar, but if I understood correctly lists always set `force_nested=true` both for serializing and deserializing. For this reason they always work.

https://github.com/robotology/yarp/blob/716b87f9975b85bf459b325e956fb86a9b38ce7a/src/idls/thrift/src/t_yarp_generator.cc#L2586-L2591

https://github.com/robotology/yarp/blob/716b87f9975b85bf459b325e956fb86a9b38ce7a/src/idls/thrift/src/t_yarp_generator.cc#L2819-L2834

So here what I see is:

1. Align `map`s to `list`s and always enable the nesting flag
1. Improve the code and check if the data type to serialize is not a base type and enable the flag if it is

I don't have the bandwidth to fill a PR complete with unit tests but I can submit a hot fix.

> Have you checked if by applying your fix, the deserialization part is correct?

From preliminary tests, editing in the same way the deserializing methods, yes.

drdanz, 2018-07-18:
> if I understood correctly lists always set force_nested=true both for serializing and deserializing

In this case I'm ok with forcing it for maps as well...

francesco-romano, 2018-07-18:
Just to reference the old issue (on which I did not provide any information): https://github.com/robotology/yarp/issues/1143

drdanz, 2018-07-19:
I closed #1143 in favour of this issue that contains a lot of extra information. Thanks for pointing that out.

closed 2018-09-14 by drdanz