robotology/yarp #1803: Import USBCamera from iCub

drdanz, 2018-07-13:


barbalberto, 2018-07-16:
Just a side note:
usbCamera device has a dependency from opencv and has some optional flags specific for iCub's event driven camera.

drdanz, 2018-07-16:
> usbCamera device has a dependency from opencv and has some optional flags specific for iCub's event driven camera.

@barbalberto Thanks for pointing that out... The OpenCV dependency is not a big deal, since we already have opencv dependencies in YARP, but I will pay particular attention to the event driven-related stuff when I perform the migration

Nicogene, 2018-07-16:
Once we import it we should expand the `usbcamera` with `macOS` and `win32` implementations:

- `macOS` : see https://github.com/ktossell/libuvc
- `win32` :  see https://github.com/ktossell/libuvc or something else if someone that uses windows knows a better library for handling usb devices (cc @randaz81, @pattacini )

drdanz, 2019-04-10:
Fixed by #1995

PR #1995 (2019-04-10): Import usbCamera driver from iCub
Fix #1803
closed 2019-04-10 by drdanz