robotology/yarp #1809: [Bindings] New template definition `Vector` is shadowing previous extensions of yarp::sig::Vector

nunoguedelha, 2018-07-20:
**Describe the bug**
We run in Matlab the following commands for creating and reading a yarp Vector:
```
>> aVec=yarp.Vector(3)

aVec = 

  3 Vector array with no properties.

>> aVec.get(0)
No appropriate method, property, or field 'get' for class
'yarp.Vector'.

Error in SwigRef/subsref (line 33)
        [varargout{1:nargout}] = builtin('subsref',self,s);
```
Initial comments [here](https://github.com/robotology-playground/sensors-calib-inertial/pull/21#issuecomment-406550545) by @lrapetti  who spotted the issue.

**To Reproduce**
Robotology-superbuild with compiled Yarp bindings.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Configuration (please complete the following information):**
 - OS: Linux or MacOS
 - yarp version: https://github.com/robotology/yarp/commit/8c421d66b57d767f7385a56fe0b7611c8c39f975
- yarp-matlab-bindings: https://github.com/robotology-playground/yarp-matlab-bindings/tree/bindings/regenerate

-  SWIG: https://github.com/robotology-dependencies/swig/tree/matlab

**Root cause**
The line https://github.com/robotology/yarp/blob/15c3328681fc8d3da5bb3bef76d686b8fbc81d64/bindings/yarp.i#L567 is shadowing the further extensions (`%extend`) of the class `yarp::sig::Vector`, such that the extra `get`, `set`, `fromMatlab`, `toMatlab` method definitions are ignored.
For more details, please refer to the analysis [here](https://github.com/robotology-playground/sensors-calib-inertial/pull/21#issuecomment-406648866).

nunoguedelha, 2018-07-20:
This issue was introduced by https://github.com/robotology/yarp/commit/2417c7d5e50aebca2ca8532fb40e7084262d50e8.
Hi @Nicogene , because of this code:
https://github.com/robotology/yarp/blob/15c3328681fc8d3da5bb3bef76d686b8fbc81d64/src/libYARP_sig/include/yarp/sig/Vector.h#L25-L39
SWIG doesn't see the `typedef` and so no `Vector` class is defined under the `yarp::sig` namespace and all lines regarding that class in `yarp.i` are ignored. Any ideas to fix this are welcome.

nunoguedelha, 2018-07-21:
Fixed by https://github.com/robotology/yarp/pull/1811.

drdanz, 2018-07-23:
Reopening since #1811 is not merged yet.
@nunoguedelha Please, do not close issues until the relative PR is merged...

closed 2018-07-23 by drdanz

nunoguedelha, 2018-07-24:
> Reopening since #1811 is not merged yet.
> @nunoguedelha Please, do not close issues until the relative PR is merged...

@drdanz sure, sorry for that.