robotology/yarp #1813: Enable -D_GLIBCXX_DEBUG and -D_GLIBCXX_DEBUG_PEDANTIC for Debug builds

drdanz, 2018-07-23:
We should enable `-D_GLIBCXX_DEBUG` and `-D_GLIBCXX_DEBUG_PEDANTIC` for Debug builds, in order to ensure that the stl usage is correct, see [this blog post](https://kristerw.blogspot.com/2018/03/detecting-incorrect-c-stl-usage.html) for futhrer details.

traversaro, 2018-07-23:
> One annoyance with the debug mode is that it changes the size of some standard class templates, so you cannot pass containers between translation units compiled with and without debug mode – this often means that you need to build the whole application with debug mode enabled.

Please, no.

drdanz, 2018-07-24:
Hmm right, I didn't read the last few lines, and I jumped to open this issue! :rofl:
Anyway, it's still an interesting option for debugging.

I also found this: https://gcc.gnu.org/onlinedocs/libstdc++/manual/debug_mode_using.html

drdanz, 2018-07-24:
Perhaps we could add it to the DebugFull builds instead...

traversaro, 2018-07-24:
> Also we don't export any artifacts, therefore I see no damage in enabling it... @traversaro what do you think?

I guess it could create problems for any test or code that use system dependencies that are compiled with vanilla libstdc++ that expose STL data structures in their public API. I think stuff that depends on PCL is probably affected, and maybe also other parts of the code.

francesco-romano, 2018-07-25:
We risk to behave like Windows! Actually even worse, as two different Debug configuration will have problems to be intermixed.
@drdanz be considerate please!

claudiofantacci, 2018-08-02:
My two cents here: I would enable those two options in two dedicated travis builds, just to understand whether we have some bad STL usage around.