robotology/yarp #1818: Asynchronous RpcClient

Nicogene, 2018-07-25:
As discussed with @drdanz, it could be nice to have a sort of `yarp::os::AsyncRpcClient`, that once he write to the `RpcServer` it return immediately a `std::future`[1].

There will be an asynchronous read callback that will valorize the `std::future` once the server answer back.

It should be quite simple to implement and nice to use it.

[1] https://en.cppreference.com/w/cpp/thread/future

pattacini, 2018-07-25:
That would be a nice addition! 👍

lornat75, 2018-07-26:
Asynch rpc isomething missing in YARP!

drdanz, 2018-07-26:
Possible implementation:

```
std::future<bool> Port::write(std::launch policy,
                              const PortWriter& writer,
                              PortReader& reader,
                              const PortWriter* callback) const
{
    std::future<bool> result = std::async(policy, [&]() {
        return write(writer, reader, callback);
    });
    return result;
}
```

swig is not happy though...

Nicogene, 2018-07-31:
[Here](https://github.com/CARVE-ROBMOSYS/Yarp-SmartSoft-Integration/blob/master/libYARP_smartsoft/include/yarp/smartsoft/QueryPatternClient.h) an implementation of something very similar to an  asynchronous RpcClient that uses the YARP ports.

drdanz, 2018-10-20:
> Possible implementation:
> 
> ```c++
> std::future<bool> Port::write(std::launch policy,
>                               const PortWriter& writer,
>                               PortReader& reader,
>                               const PortWriter* callback) const
> {
>     std::future<bool> result = std::async(policy, [&]() {
>         return write(writer, reader, callback);
>     });
>     return result;
> }
> ```

We discussed this, and it is definitely not the right way to implement this, since this will start a thread every time. An alternative is to use [`std::packaged_task`](https://en.cppreference.com/w/cpp/thread/packaged_task).

Nicogene, 2018-10-23:
I think that there is another problem, unfortunately the output (the reply of the rpc request) is returned by reference and there is no way to "lock"   the `PortReader`, the user can modify it while the other thread is working on it.

Since it will be a new method, we could think to return the reply in a `std::pair<PortReader, bool>` moving to [`std::optional`](https://en.cppreference.com/w/cpp/utility/optional) when we will move to c++ 17