robotology/yarp #1826: yarp bindings give error with Vector conversion

towardthesea, 2018-08-06:
**Describe the bug**
I used to do following in `Python` with `yarp bindings` to convert a `Swig object of Vector` to `Vector` in `yarp 2.x`, for example when using `iKin` to get the current joint angles of the `iCubArm`
```
        angle = self._iArm.getAng()  #_iArm = icub.iCubArm('right')
        print('angle', angle)
        angle_vector = yarp.Vector(angle)  # my result yarp Vector of joint angles
```
However, I got following error when doing the same with `yarp 3.x`

```
('angle', <Swig Object of type 'yarp::sig::Vector *' at 0x7f82a0098240>)
Traceback (most recent call last):
  File "demo/yarp_demo.py", line 550, in <module>
    arm_estimation.runModule(rf)
...
    return _yarp.RFModule_runModule(self, *args)
  File "demo/yarp_demo.py", line 264, in updateModule
    self.predict_yarp
  File "demo/yarp_demo.py", line 443, in predict_yarp
    angle_vector = yarp.Vector(angle)
  File "/home/pnguyen/icub-workspace/yarp/build/lib/python/yarp.py", line 13830, in __init__
    this = _yarp.new_Vector(*args)
NotImplementedError: Wrong number or type of arguments for overloaded function 'new_Vector'.
  Possible C/C++ prototypes are:
    yarp::sig::VectorOf< double >::VectorOf()
    yarp::sig::VectorOf< double >::VectorOf(size_t)
    yarp::sig::VectorOf< double >::VectorOf(size_t,double const &)
    yarp::sig::VectorOf< double >::VectorOf(size_t,double const *)
    yarp::sig::VectorOf< double >::VectorOf(yarp::sig::VectorOf< double > const &)

swig/python detected a memory leak of type 'yarp::sig::Vector *', no destructor found.
```
and the application failed.

**Configuration (please complete the following information):**
 - OS: ubuntu 16.04
 - yarp version: >3.1 (devel)
-  compiler:

Probably some changes with the definition of `Vector` in `yarp 3.x` causes some mismatches after conversion of `swig`.

towardthesea, 2018-08-06:
cc @nunoguedelha @traversaro

towardthesea, 2018-08-07:
Add another `typedef` in `yarp.i` file like this:
```
%inline %{
    typedef yarp::sig::VectorOf<double> Vector;
%}
```
before:
```
%template(Vector) yarp::sig::VectorOf<double>;
```
can help solve the issue.

Tested with **python binding** only!

nunoguedelha, 2018-08-07:
Very nice solution! hoping it will work with Matlab bindings. Well it should. I'll test ASAP.

towardthesea, 2018-08-07:
Also worked with Java through Matlab (I mean [this](http://www.yarp.it/yarp_swig.html#yarp_swig_matlab) )!

nunoguedelha, 2018-08-07:
I just tested:
- the Matlab bindings using the Mex file directly still work fine
- I couldn't get rid of the fix that replaced all `yarp::sig::Vector` instances by `yarp::sig::VectorOf` (https://github.com/robotology/yarp/pull/1811)

Beware of this issue https://github.com/robotology/yarp/commit/2417c7d5e50aebca2ca8532fb40e7084262d50e8, which fix consisted in making SWIG skip the `typedef` of `Vector`: https://github.com/robotology/yarp/blob/2417c7d5e50aebca2ca8532fb40e7084262d50e8/src/libYARP_sig/include/yarp/sig/Vector.h#L35-L37

Without this guard, the Ruby bindings generation were crashing!

CC @Nicogene

towardthesea, 2018-08-08:
I can compile with `Lua` and `Ruby` options but cannot test them. Maybe @vtikha  can help with `Lua test`

nunoguedelha, 2018-08-09:
Python bindings fixed by https://github.com/robotology/yarp/pull/1828. Also fixes some left overs on Matlab bindings regarding issue with `yarp::sig::Vector`.

traversaro, 2018-08-21:
Should have been fixed by the merge of https://github.com/robotology/yarp/pull/1828 . Let us know if there are more problems.

Nicogene, 2018-08-29:
I removed 
```
inline %{
typedef yarp::sig::VectorOf<double> Vector;
%}
```
from `yarp.i`, keeping `%template(Vector) yarp::sig::VectorOf<double>;`
and the test proposed by @traversaro 
```python
import yarp
import unittest

class VectorTest(unittest.TestCase):

    def test_vector_copy_costructor(self):
        vecSize = 10
        vec = yarp.Vector(vecSize)
        self.assertEqual(vec.size(), vecSize)
        vecCheck = yarp.Vector(vec)
        self.assertEqual(vec.size(), vecCheck.size())

if __name__ == '__main__':
    unittest.main() 

```
works fine.
Maybe we are missing the point here, what is the type of `angle`?

nunoguedelha, 2018-08-29:
FYI there was a typo in https://github.com/robotology/yarp/pull/1828, so I've submitted a PR https://github.com/robotology/yarp/pull/1840 with what should be a cleaner fix, not using `%inline` at all but `%{...%}` along with the `%template` which was already there.

towardthesea, 2018-08-29:
> Actually the test that I proposed was not effective in capturing the issue. I guess it is something related to functions that are returing a yarp::sig::Vector, but I could not find one to test in YARP that is wrapped in the bindings.

@Nicogene  Indeed, the errors with `VectorOf` only show when you call a function with `Vector` as argument.

Nicogene, 2018-08-29:
Ah I see! The problem wasn't the copy constructor but this line:

```python
print('angle',angle)
```
where `angle` is a yarp::sig::Vector.

Nicogene, 2018-08-29:
The problem comes out when you try to construct a Vector from a Vector returned by a function, in #1840 I added test that check this issue.

Nicogene, 2018-08-29:
Closing until #1840 will be merged