robotology/yarp #1827: IDL generated files have IIT copyright

Tobias-Fischer, 2018-08-07:
**Describe the bug**
Files that are generated using thrift / IDL are under copyright for IIT under BSD-3.

**To Reproduce**
Generate .h/.cpp/.cmake files using a .thrift file and the `yarp_idl_to_dir` cmake command. An example can be found [here](https://github.com/robotology/icub-hri/tree/master/src/modules/iol2opc)

**Expected behavior**
One should be able to specify the license and copyright holder. In the specific example, it should be `WYSIWYD Consortium` and `GPL v.2.0`.

**Configuration (please complete the following information):**
 - yarp version: >=3.0

**Additional context**
While for many cases within the `yarp` and `icub-main` directories it is probably okay to have the default IIT copyright header, in the specific example I just encountered the wrong copyright was added. Just because the tool was developed by IIT, it does not mean that the files themselves are under the same copyright (as far as I know, at least). In yarp<3, the files did not have a copyright header at all. I noticed this changed in my recent updates for the `icub-hri` repository: https://github.com/robotology/icub-hri/commit/37168d6f86fecc803576b5ef34640c8ab4623e5a

It would be great if one could somehow specify the copyright header one wants to use.

drdanz, 2018-08-07:
I don't think this is a real issue, the tool simply produces files that can be modified/re-distributed according to the BSD-3 license. If you want to modify them, and re-distribute them with a more restrictive license, you are free to do it, as long as you follow the BSD-3 requirements.

AFAIK, this is not uncommon for tools generating code, see for example Bison that produces code that is GPL-3+, with a special exception, copyright Free Software Foundation, Inc.

Also maybe unrelated, but from [GPL FAQs](https://www.gnu.org/licenses/old-licenses/gpl-2.0-faq.en.html#WhatCaseIsOutputGPL)

>Q:In what cases is the output of a GPL program covered by the GPL too?
>
>A: Only when the program copies part of itself into the output.

@lornat75 what do you think?

drdanz, 2021-02-01:
The `yarpidl_thrift` tool has a `no_copyright` switch that can be enabled to skip the generation of the copyright headers.
I think this issue can be closed.

closed 2021-02-01 by drdanz