robotology/yarp #1829: System-wide libedit not found

claudiofantacci, 2018-08-12:
**Describe the bug**
macOS comes with a system-wide `libedit` located in `/usr/local` that is usually found by CMake and used, e.g., to enhance the UX of YARP RPC ports (command line like input, tokenization, etc.). System-wide `libedit` is no longer found by CMake and the only way to get it trough is to install libedit with brew and to force link it `brew install libedit && brew link --force libedit`.

**To Reproduce**
Configure YARP with **Xcode version 9+**.
I'm not sure how Xcode actually as an impact on this, and in particular with CMake, but this correlation is also confirmed by our Travis builds:
 - **Master**
   - [Compiled with Xcode 8.3](https://travis-ci.org/robotology/yarp/jobs/413038983#L2934) - Found
   - [Compiled with Xcode 9.4](https://travis-ci.org/robotology/yarp/jobs/413038984#L1188) - Not found
 - **Devel**
   - [Compiled with Xcode 8.3](https://travis-ci.org/robotology/yarp/jobs/413508032#L2934) - Found
   - [Compiled with Xcode 9.4](https://travis-ci.org/robotology/yarp/jobs/413508033#L1196) - Not found

**Expected behavior**
System-wide `libedit` found without the need to force link `libedit` under `brew`.

**Configuration (please complete the following information):**
 - OS: `10.13.4`
 - yarp version: `3.1.100+20180802.3+git013120125`
-  compiler: Apple LLVM version `9.0.0` (`clang-900.0.39.2`)

**Additional context**
My only guess so far is that by upgrading to Xcode 9+ the (possibly present, I can't check it) pkg-config `libedit.pc` has been removed and the [`standard_find_module`](https://github.com/robotology/ycm/blob/master/modules/StandardFindModule.cmake) (invoked [here](https://github.com/robotology/ycm/blob/master/find-modules/FindLibedit.cmake#L44)) is no longer able to find the library. As a matter of the fact, by force linking `libedit` with `brew`, `standard_find_module` indeed finds the `libedit.pc` that is shipped with it.

**Additional comments**
As a matter of the fact/consequence, the Travis command related to [`brew install libedit`](https://github.com/robotology/yarp/blob/master/.travis.yml#L358) is useless unless it is `brew link --force libedit`. By checking the Travis log it is clearly reported that the formula is `keg-only`, i.e. it is not symlinked because libedit is shipped with the OS.

francesco-romano, 2018-08-13:
Hi @claudiofantacci 
the `.pc` file has been removed.. I actually triggered the removal (bug #28114484 on Apple bug reporters) as it was wrong anyway before. 
Anyway, the `.pc` should not be necessary as the lib should be in the system path, and hopefully, the `find_package` looks there.

I think the problem you have is that you do not have the command line tools installed. 
Try to run  `xcode-select --install`: this should symlink the header files from `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/` to `/usr/`

claudiofantacci, 2018-08-13:
Hi @francesco-romano, thanks for your comment. Now I know is your fault 😛 ! j/k 😉 

I already `xcode-select --install`ed, so the problem lies somewhere else.
On YARP the `standard_find_module`'s `find_package(libedit NO_MODULE)` does not find the library and it falls back on using `pkg_config`, you can try configuring with `cmake <path> --trace` and see it.

francesco-romano, 2018-08-13:
Can you check you have the headers in `/usr/include` ?

```
fraromano-macbookpro:~ fraromano$ ls -l /usr/include/ | grep edit
drwxr-xr-x    3 root  wheel       96 13 Aug 10:53 editline
-rw-r--r--    1 root  wheel     9246  7 Oct  2017 histedit.h
fraromano-macbookpro:~ fraromano$ ls -l /usr/include/editline/
total 0
-rw-r--r--  1 root  wheel  7459  7 Oct  2017 readline.h
```

claudiofantacci, 2018-08-13:
Yep!

```bash
Claudio @ MBP-di-Claudio ~ 12:01 pm
└─ ~
└─ $ ls -l /usr/include/ | grep edit
drwxr-xr-x    3 root  wheel    96B 15 Lug  2017 editline/
-rw-r--r--    1 root  wheel   9,0K 15 Lug  2017 histedit.h

Claudio @ MBP-di-Claudio ~ 12:01 pm
└─ ~
└─ $ ls -l /usr/include/editline/
total 0
drwxr-xr-x    3 root  wheel    96B 15 Lug  2017 ./
drwxr-xr-x  263 root  wheel   8,2K  6 Mar 12:04 ../
-rw-r--r--    1 root  wheel   7,3K 15 Lug  2017 readline.h
```

francesco-romano, 2018-08-13:
Ok. I think it is a problem of the `FindLibedit.cmake` then. We should check if it tries to look for `histedit.h` in `/usr/local/include` or not.

francesco-romano, 2018-08-13:
Is yarp using the YCM `FindLibedit.cmake`? 
Maybe @drdanz can help on this.

claudiofantacci, 2018-08-13:
> Is yarp using the YCM FindLibedit.cmake?

Yes, the call trace is reported in **Additional context** of my first post 😄 

> Just to be clear.. I think we need something like `find_path` and `find_library` hinting to `/usr/include` and `/usr/lib`

The point is that there is no `FindLibedit.cmake` so `pkg_config` is always used and was also used until Xcode 9+. If we don't want to use `pkg_config` we need a dedicated `FindLibedit.cmake`.

francesco-romano, 2018-08-13:
Ok, I might have missed a bit the "stack" for the cmake calls.

Then, yes. I can confirm that the `.pc` has been removed as I triggered it 🤣 
IMHO there is no need to use the `brew` version, but we should create the `FindLibedit.cmake` to handle the case where the system provides the library.

Nicogene, 2018-08-29:
What do you think about moving this issue in YCM?
I think that the fix should be done there and then imported in YARP.

claudiofantacci, 2018-08-29:
I'm ok with that.

Nicogene, 2018-08-29:
moved to robotology/ycm#166

closed 2018-08-29 by Nicogene

andlabs, 2019-03-28:
@francesco-romano Would you be willing to copy the contents of Apple bug 28114484 to OpenRadar? Apple bugs are not visible to anyone except the person who filed the bug, and the missing pkg-config file is currently confusing a few other people in an unrelated IRC channel.