robotology/yarp #1831: YCM CMakeLists.txt uses wrong minimum version for installing cmake-next

Tobias-Fischer, 2018-08-15:
**Describe the bug**
The `cmake-next` folder within ycm contains patches and modifications that are to be merged into `cmake` (see https://github.com/robotology/ycm/tree/master/cmake-next). In the CMakeLists.txt for YCM that is contained within YARP, version 3.8 is hardcoded as "stable" version (as far as I can tell): https://github.com/robotology/yarp/blob/master/extern/ycm/CMakeLists.txt#L70-L75
Therefore, the cmake-next modules do not get installed when one has cmake > 3.8 (for example, I have version 3.11). I'd like to use the FindPython2 / FindPython3 modules, which were added in version 3.12 and are contained within ycm/cmake-next, however they don't get installed because of the hardcoded 3.8

**To Reproduce**
Do not manually install ycm (let yarp do the job). Use cmake versions >3.8. make & make install yarp.

**Expected behavior**
I expect the following folder to be created: `$YARP_DIR/share/yarp/cmake/ycm-0.9.0/cmake-next`.

**Configuration (please complete the following information):**
 - OS: Linux
 - yarp version: 3.1.0

**Additional context**
The bug report is opened here as the top-level CMakeLists.txt for YCM differs between the one provided here in YARP (https://github.com/robotology/yarp/blob/master/extern/ycm/CMakeLists.txt) and the one directly in YCM (https://github.com/robotology/ycm). I guess keeping them synchronous in the long-term should be tackled somehow as well.

In https://github.com/robotology/icub-hri I'd like to use the FindPython2 and FindPython3 modules from CMake 3.12. Using the ones provided by YCM would be optimal, rather than having to copy them myself into iCub-HRI.

Once the bug described here is fixed, I've got another question. If I want to use YCM contained in YARP, I'd need to use `YARP_MODULE_PATH`, which is deprecated. What is the long-term vision to resolve this? Will YCM become a hard dependency?

PeterBowman, 2018-08-15:
If I'm correct, the YCM copy shipped within YARP is just a subset of the *full* YCM project. That is, it contains everything you'll need to build YARP and nothing more. In order to fully benefit from its features, you should install YCM as a standalone package on your system (and let YARP use that instead of `extern/ycm/`).

> What is the long-term vision to resolve this? Will YCM become a hard dependency?

It's already becoming a hard dep in some robotology-related repo, check the comments at https://github.com/robotology/yarp/issues/1777.

Tobias-Fischer, 2018-08-16:
I see. For icub-hri (aimed at newcomers as well), I'd not like to make YCM a dependency until it is not a dependency for icub-main (see https://github.com/robotology/icub-main/issues/534). I'll keep an eye on that issue.
In the meantime, I think the CMakeLists.txt here in yarp should still be fixed. If someone explains the logic behind cmake-next, cmake-proposed etc I can have a go. I am also a little confused what is considered cmake-next within ycm, as it seems that 3.12 is considered cmake-next while it was released already.

drdanz, 2018-09-06:
YCM files installed by YARP are there just for historical reason, only the files that were originally installed before we decided to start developing YCM are still installed. They are not meant to be used by any new project, their use is deprecated and they are already not installed when `YARP_NO_DEPRECATED` is enabled. Also they should have been removed completely in YARP 3, but in the end we left them because this has a major impact on the robot setup, but they **will be removed soon**. I'm not going to accept any patch that fosters the usage of the YCM files installed by YARP, sorry.

closed 2018-09-06 by drdanz