robotology/yarp #1843: YARP ccmake error when RTF_ADDONS = ON

samialperen, 2018-08-31:
I have installed Robot Testing Framework (RTF) from github page https://github.com/robotology/robot-testing and i followed instructions to install. It is compiled successfully. I cloned it to my home directory so at at the end my build directory is /home/username/mydirectory/robot-testing/build 

I configured the environment to set RTF_DIR to: /home/username/mydirectory/robot-testing/build 
and my PATH accordingly. When i write echo $RTF_DIR i can see the output. However, when i try to compile yarp with ccmake while YARP_COMPILE_RTF_ADDONS = ON it gives the following error:

` CMake Error at cmake/YarpFindDependencies.cmake:164 (message):
   Optional package RTF not found.  Please install it or disable the option
   "YARP_COMPILE_RTF_ADDONS" to build yarp.
 Call Stack (most recent call first):
   cmake/YarpFindDependencies.cmake:521 (check_optional_dependency)
   CMakeLists.txt:63 (include)`

However, when it is off, i can compile YARP successfully and there is no problem. Apparently it can not find RTF_DIR. Indeed, in ccmake window it says RTF_DIR not found.

Can somebody help me where is the problem?

traversaro, 2018-08-31:
Hi @samialperen , unfortunately YARP 3.1.0 depends on RTF 1.5.0, that is currently unreleased, see https://github.com/robotology/robot-testing/issues/99 . Sorry for the trouble.

For the time being, if you manually build `RTF` you can just switch to the `devel` branch of RTF to build and install the RTF 1.5.0 . 

If you want to make sure to always use compatible version of the robotology libraries, you can consider using the [`robotology-superbuild`](https://github.com/robotology/robotology-superbuild): both YARP and RTF are part of the libraries installed by default in the `robotology-superbuild`.

samialperen, 2018-08-31:
@traversaro Thank you very much. I installed RTF-devel instead of RTF-master and it is worked now!

samialperen, 2018-08-31:
Hey @traversaro , i am following vvv-school 18 tutorials and unfortunately i have another problem with RTF & YARP now. When i try to run smoke-test/test.sh files i have the following error:

`[testrunner] cannot load plugin yarpmanager; error (foun) : yarpmanager.so: cannot open shared object file: No such file or directory
./helpers/scripts/smoke-test-cpp.sh: line 100:  1190 Killed                  yarprun --server /testnode --log > /dev/null 2>&1
`

I have checked issues and it seems you had same problem before as well. Could you help me about this?

traversaro, 2018-09-01:
Hi @samialperen, did you appended the `<prefix>/lib/rtf` directory to `LD_LIBRARY_PATH` environmental variable (where `<prefix>` is the installation prefix of YARP), as documented in http://www.yarp.it/rtf_plugins.html ? From your question I guess there is probably some missing docs on this in the VVV setup instructions. 

Related issues/PR: 
* https://github.com/robotology/yarp/issues/1375
* https://github.com/robotology/yarp/pull/1443

pattacini, 2018-09-01:
[VVV18 installation instructions][1] should be up-to-date

As @traversaro mentioned, most of the YARP tutorials still require an update if you're relying on the latest YARP release. Hoewver, if you like, there's the possibility to use a VM with everything needed to follow the VVV18 tutorials pre-installed: [check it out][2]

[1]: https://github.com/vvv-school/vvv18/wiki/How-to-prepare-your-system
[2]: https://github.com/vvv-school/vvv18/wiki/Before-arriving-at-VVV

pattacini, 2018-09-01:
Hold on @samialperen the VM does not contain the snapshots of the repositories as used at vvv18. We're going to deliver git tags for that.

samialperen, 2018-09-01:
Hey @pattacini @traversaro 
I have tried yarp master branch with ccmake RTF_ADDONS = ON and i used sudo make install. In this case yarp installation directory is /usr/local automatically. 
I used sudo make install for RTF as well. When i navigate to /usr/local/lib/rtf directory i can see yarpmanager.so, yarpplugin.so and yarpserver.so. So they are installed correctly. YARP works quite fine when RTF_ADDONS = OFF. I added the following to my .bashrc:
 `export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/usr/local/lib:$/usr/local/lib/rtf`
But still same problem exists. I tried with yarp-devel branch and it is the same. What i am missing? :(

traversaro, 2018-09-01:
Have you opened a new terminal/sourced `.bashrc` after the modification?
In the terminal where you have the problem, can you make sure that `LD_LIBRARY_PATH` has the intended value and `/usr/local/lib/rtf` actually contains the `yarpmanager.so` library? 
Which operating system are you using?

samialperen, 2018-09-01:
Yes i did. Here is some steps i followed in the new terminal:
```
echo $LD_LIBRARY_PATH
/home/alperen/MY_WORKSPACES/catkin_ws/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/usr/local/lib:$/usr/local/lib/rtf
```
```
ls /usr/local/lib/rtf
yarpmanager.so  yarpplugin.so  yarpserver.so
```
```
cd assigment_motor_control/smoke-test
sudo rm -rf build
sudo ./test.sh
```
But still the same error.

samialperen, 2018-09-01:
And here is the full output of test.sh:

```
Cloning into 'helpers'...
remote: Counting objects: 17, done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 17 (delta 2), reused 4 (delta 0), pack-reused 0
Unpacking objects: 100% (17/17), done.
Checking connectivity... done.
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Libraries are placed in /home/alperen/Programs/Imperial/Tutorials/iCub_Tutorials/assignment_motor-control/smoke-test/build/build-code/lib
-- Executables are placed in /home/alperen/Programs/Imperial/Tutorials/iCub_Tutorials/assignment_motor-control/smoke-test/build/build-code/bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alperen/Programs/Imperial/Tutorials/iCub_Tutorials/assignment_motor-control/smoke-test/build/build-code
Scanning dependencies of target trigger
[ 16%] Building CXX object CMakeFiles/trigger.dir/src/trigger.cpp.o
[ 33%] Linking CXX executable bin/trigger
[ 33%] Built target trigger
Scanning dependencies of target server
[ 50%] Building CXX object CMakeFiles/server.dir/src/server.cpp.o
[ 66%] Linking CXX executable bin/server
[ 66%] Built target server
Scanning dependencies of target client
[ 83%] Building CXX object CMakeFiles/client.dir/src/client.cpp.o
[100%] Linking CXX executable bin/client
[100%] Built target client
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/bin/server
-- Set runtime path of "/usr/local/bin/server" to "/usr/local/lib"
-- Installing: /usr/local/bin/client
-- Set runtime path of "/usr/local/bin/client" to "/usr/local/lib"
-- Installing: /usr/local/bin/trigger
-- Set runtime path of "/usr/local/bin/trigger" to "/usr/local/lib"
-- Installing: /usr/local/share/ICUBcontrib/applications/assignment_motor-control-system.xml
-- Installing: /usr/local/share/ICUBcontrib/applications/assignment_motor-control.xml
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alperen/Programs/Imperial/Tutorials/iCub_Tutorials/assignment_motor-control/smoke-test/build/build-test
Scanning dependencies of target TestAssignmentMotorControl
[ 50%] Building CXX object CMakeFiles/TestAssignmentMotorControl.dir/test.cpp.o
[100%] Linking CXX shared module plugins/TestAssignmentMotorControl.so
[100%] Built target TestAssignmentMotorControl
Looking for name server on 146.169.184.82, port number 10000
If there is a long delay, try:
  yarp conf --clean
Name server /root is available at ip 146.169.184.82 port 10000
Loading /home/alperen/Programs/Imperial/Tutorials/iCub_Tutorials/assignment_motor-control/smoke-test/test.xml
[testrunner] cannot load plugin yarpmanager; error (foun) : yarpmanager.so: cannot open shared object file: No such file or directory
Scanning dependencies of target uninstall
-- Uninstalling "/usr/local/bin/server"
-- Uninstalling "/usr/local/bin/client"
-- Uninstalling "/usr/local/bin/trigger"
-- Uninstalling "/usr/local/share/ICUBcontrib/applications/assignment_motor-control-system.xml"
-- Uninstalling "/usr/local/share/ICUBcontrib/applications/assignment_motor-control.xml"
Built target uninstall
./helpers/scripts/smoke-test-cpp.sh: line 107: 26816 Killed                  yarprun --server /testnode --log > /dev/null 2>&1  (wd: ~/Programs/Imperial/Tutorials/iCub_Tutorials/assignment_motor-control/smoke-test/build)
Unable to get test result

xxxxx Test FAILED xxxxx
```

pattacini, 2018-09-01:
@samialperen make the attempt of launching the test without `sudo`, starting from a clean condition.

samialperen, 2018-09-01:
@pattacini  After removed smoke-test/build, i executed directly ./test.sh without sudo and here is the error i got:
```
[ 16%] Building CXX object CMakeFiles/trigger.dir/src/trigger.cpp.o
[ 33%] Linking CXX executable bin/trigger
[ 33%] Built target trigger
Scanning dependencies of target server
[ 50%] Building CXX object CMakeFiles/server.dir/src/server.cpp.o
[ 66%] Linking CXX executable bin/server
[ 66%] Built target server
Scanning dependencies of target client
[ 83%] Building CXX object CMakeFiles/client.dir/src/client.cpp.o
[100%] Linking CXX executable bin/client
[100%] Built target client
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/bin/server
-- Set runtime path of "/usr/local/bin/server" to "/usr/local/lib"
-- Installing: /usr/local/bin/client
-- Set runtime path of "/usr/local/bin/client" to "/usr/local/lib"
-- Installing: /usr/local/bin/trigger
-- Set runtime path of "/usr/local/bin/trigger" to "/usr/local/lib"
-- Installing: /usr/local/share/ICUBcontrib/applications/assignment_motor-control-system.xml
CMake Error at cmake_install.cmake:96 (file):
  file INSTALL cannot copy file
  "/home/alperen/Programs/Imperial/Tutorials/iCub_Tutorials/assignment_motor-control/app/scripts/assignment_motor-control-system.xml"
  to
  "/usr/local/share/ICUBcontrib/applications/assignment_motor-control-system.xml".

Makefile:61: recipe for target 'install' failed
make: *** [install] Error 1

```

samialperen, 2018-09-01:
@pattacini Unfortunately the problem is not about icub-contrib-common. assignment_inverse_kinematics does not use icub and when i try to run it without sudo i still have yarpmanager.so error.

pattacini, 2018-09-01:
@samialperen `assignment_inverse-kinematics` [does use it][1].
However, this doesn't mean that the problem is related to that.

I've got a personal server on the cloud where main projects (e.g. `yarp`, `icub-main`) are installed in `/usr/local` whereas `ICUBcontrib_DIR` points to `/home/...` path and everything works just fine.

On the other hand, [VVV18 instructions][2] suggests using local paths (e.g. `~/robot-install`) for installation. This is the way the VM is prepared and I would recommend you to give it a whirl.

As of now, I don't have any system configured as yours for doing a test, where both main projects and `icub-contrib-common` are installed in `/usr/local`.

[1]: https://github.com/vvv-school/assignment_inverse-kinematics/blob/master/CMakeLists.txt#L12
[2]: https://github.com/vvv-school/vvv18/wiki/How-to-prepare-your-system

samialperen, 2018-09-01:
@pattacini and @traversaro thank you very much for help. As you mentioned, the problem occurs when you try to install default location (/usr/local). I installed everything under a local path and now it works perfectly.

closed 2018-09-01 by samialperen