robotology/yarp #1846: YARP uses non-locale safe function `std::atof`

traversaro, 2018-09-03:
YARP uses in several point of the codebase the non-locale safe function `std::atof`, see https://github.com/robotology/yarp/search?utf8=%E2%9C%93&q=atof&type=. This means that the parsing of configuration files or parameters can depend on the global locale settings of the process in which YARP is used, resulting in silent failures in reading floating point numbers that may be quite difficult to debug. See https://github.com/robotology/idyntree/issues/288 and the links for several examples of this problem. 

YARP should parse its configuration files always in a consistent way, regardless of the process or system locale. Possible solution involve using the `stringstream` libraries with the `imbue` method ( as in https://github.com/robotology/idyntree/blob/114671fb1a012f6214b35b8462d463fea99781fd/src/model_io/urdf/include/private/URDFParsingUtils.h#L29 ) or using custom conversion functions.

drdanz, 2018-09-06:
> This means that the parsing of configuration files or parameters can depend on the global locale settings of the process in which YARP is used

Are you sure about this? if I remember correctly parsing the .ini files ends up in

https://github.com/robotology/yarp/blob/bc6b07c7a86b366adcec635d919a759a7972dc18/src/libYARP_OS/src/Storable.cpp#L435:L438

and in the end in

https://github.com/robotology/yarp/blob/bc6b07c7a86b366adcec635d919a759a7972dc18/src/libYARP_OS/src/Storable.cpp#L108:L119

That explicitly handles the locale...

Perhaps the problem with the config files is in yarpmanager only?

The other instances that I see, doesn't seem to have a heavy impact... I don't think that anyone ever used the `NONSENSE_ADMIN_DELAY` in `PortCore`, the frequency reported by SystemInfo is probably wrong, but not actually used anywhere afaik

traversaro, 2018-09-06:
> Perhaps the problem with the config files is in yarpmanager only?

Exactly.

traversaro, 2018-09-06:
I think also the one in Map2DServer is relevant: https://github.com/robotology/yarp/blob/f5e903f58cd4550504d6f6643e0bb3f91ece2ffe/src/devices/map2DServer/Map2DServer.cpp#L736 .

drdanz, 2018-09-06:
I think we should move the `fp_from_string()` method in [NetType](https://github.com/robotology/yarp/blob/bc6b07c7a86b366adcec635d919a759a7972dc18/src/libYARP_OS/include/yarp/os/NetType.h), it looks like there are already related methods there...