robotology/yarp #1850: Failure to open a new remotecontrolboard client in simulink

Yeshasvitvs, 2018-09-05:
**Describe the bug**
In simulink I have a main `config` block with all the control boards i.e., `{'torso','left_arm','right_arm','left_leg','right_leg'}`. Then I have a subsystem with a new config block containing only a subset of the control boards i.e., `{'torso','left_arm','right_arm'}`. I used different prefixes for both the config blocks. Now, when I try to run the controller it terminates with the following error:

yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
yarpdev: ***ERROR*** driver <remotecontrolboardremapper> was found but could not open

when trying to open the new client for the subsystem

If the subsystem is commented out, the model runs without any errors 

NOTE: This behavior is observed with icubGazeboSim and is not checked on the real robot.

**To Reproduce**
Create a simple simulink model with the name `test.mdl` and with a config block and any simple get measurements block as shown below. Set the `Local Name` under config block as `WBT`

[image]

Now, create a subsystem within this model and use another config block with a subset of control boards under the `control board names` tab and set the `Local Name` as `WBT_SUB`. Place either a get measurements or set measurements block as shown below

[image]

Now, when you run the controller, you will see the following error

yarp: Port /WBT_SUB/testRobotModel
PositionControlConfiguration/icubSim/torso/rpc:o active at tcp://10.255.35.136:10238/
yarp: Port /WBT_SUB/testRobotModel
PositionControlConfiguration/icubSim/torso/command:o active at tcp://10.255.35.136:10239/
yarp: Port /WBT_SUB/testRobotModel
PositionControlConfiguration/icubSim/torso/stateExt:i active at tcp://10.255.35.136:10240/
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
yarpdev: ***ERROR*** driver <remotecontrolboardremapper> was found but could not open
Failed to open the RemoteControlBoardRemapper with the options passed.
Failed to initialize the RemoteControlBoardRemapper.
Failed to get IEncoders interface.

**Expected behavior**
Ideally, a new controlboard client should open for the subsystem with the port prefix `WBT_SUB` and connect to the robot ports without error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Configuration (please complete the following information):**
 - OS: Link 
 - yarp version: 3.1.0+10-20180821.4+gitd661b836b
-  compiler: gcc

**Additional context**
Add any other context about the problem here.

@diegoferigo Please add any missing details. Thanks

traversaro, 2018-09-05:
@Yeshasvitvs have you tried to check out that adding a delay in the port opening is solving the problem? Otherwise blaming YARP may be a bit premature.

diegoferigo, 2018-09-05:
@traversaro Yes he added a delay and it did not solve it.

As far as I understood this is related to the sequential opening of two `remotecontrolboard`s (through the remapper) in the case they share one or more control boards. Under these circumstances, the second one would fail to open complaining about the access to the ports of the wrapper.

@Yeshasvitvs Could you please try what follows?

1. Provide the output you have in the Matlab prompt. I remember it was more verbose than the one you posted.
1. Does this happen also if the two instances do not share any control board?

Yeshasvitvs, 2018-09-05:
### Error from matlab diagnostic viewer while running test.mdl

[image]

@diegoferigo While using the simple model `test.mdl` I do not have any errors on the matlab prompt. The errors you were referring to happened while running on the pHRI standup model.

Yeshasvitvs, 2018-09-05:
@diegoferigo I tried to run the `test.mdl` with the main config block containing only the leg control boards and the config block of the subsystem same as before. It gives the same error as failed to open the controlboard client for the subsystem!

Nicogene, 2018-09-06:
Ping @barbalberto and/or @randaz81.
BTW, why this warning:
```
[WARNING] Updating to newer yarp and the usage of controlBoardWrapper2 is suggested***
```
you are using the newest yarp (3.1.0), are you using `ControlBoardWrapper` instead of `ControlBoardWrapper2`?
There is a reason for that?
I don't know the history behind the two wrappers, who's is better to use, but maybe the old one is outdated/deprecated.

diegoferigo, 2018-09-06:
@Nicogene The source should be this:

https://github.com/robotology/yarp/blob/f5e903f58cd4550504d6f6643e0bb3f91ece2ffe/src/libYARP_dev/src/devices/RemoteControlBoard/RemoteControlBoard.cpp#L1102-L1118

traversaro, 2018-09-06:
@Nicogene That error is misleading: there is a failure in connecting to the `stateExt:o` port, and so the `remote_controlboard` is assuming that the user is trying to connect to a legacy `controlboarwrapper`. If `controlboardwrapper` still exist, that error should be printed only if `connectionProblem` is not set to true.

barbalberto, 2018-09-17:
I don't know how the code is structured in matlab, my first guess is that the `Network` object is no longer availalble where the second set of controlboard are created. Can you add the following check right before creating them?

```
    if (!yarp.checkNetwork())
    {
        printf("No yarp network, quitting\n");
        return 1;
    }
```