robotology/yarp #1853: Network.h contains method that should be somewhere else

Nicogene, 2018-09-12:
`Network` for yarp have ever been a *superclass* that does a lot of stuff, some of them not directly related to the **network** or communication.

Giving a look to `Network.h` I found that these methods:
```c++
    /**
     *
     * Read a variable from the environment.
     *
     * @param key the variable to read
     * @param found an optional variable to set to true iff variable is found
     * @return the value of the environment variable, or "" if not found
     *
     */
    static std::string getEnvironment(const char *key,
                                      bool *found = nullptr);

    /**
     *
     * Set or change an environment variable.
     *
     * @param key the variable to set or change
     * @param val the target value
     *
     */
    static void setEnvironment(const std::string& key, const std::string& val);

    /**
     *
     * Remove an environment variable.
     *
     * @param key the variable to remove
     *
     */
    static void unsetEnvironment(const std::string& key);

    /**
     *
     * Get an OS-appropriate directory separator (e.g. "/" on linux)
     *
     */
    static std::string getDirectorySeparator();

    /**
     *
     * Get an OS-appropriate path separator (e.g. ":" on linux)
     *
     */
    static std::string getPathSeparator();
```

should stay somewhere else, in a sort of utility header, what do you think about it?

This should make `Network` a lighter class, that does only what it have to do, making it easier to handle and maintain.

This is not a blocking issue, but I think it is something to not ignore because this(like other cases) causes "cross-dependencies" because some things are not in the right place.
For example see this [comment](https://github.com/robotology/yarp/pull/1852#discussion_r216931763)

Nicogene, 2019-12-04:
@drdanz shall we move in `yarp::conf::environnment` the `setEnvironment` and `unsetEnvironment`?

drdanz, 2019-12-04:
Yes, but before doing that I'd like to find out if there is some proposal to the standard, or something similar. There are already `std::setenv` and `std::getenv` in `cstdlib`, but I'd like to investigate if there is any plan for c++-like methods

closed 2020-07-29 by drdanz

drdanz, 2020-07-29:
Fixed by #1855 and #2334

PR #2334 (2020-07-29): [Discussion] Add yarp::conf::environment::{get,set,unset}Environment() functions and deprecate NetworkBase version and yarp::os::getenv
This is the alternative version for #2333 which moves these functions in the `yarp::conf::environment` namespace.

Please keep the discussion on #2333

---------

### Libraries

#### `conf`

* Added the `yarp/conf/environment.h` header.
* Added the following functions:
  * `yarp::conf::environment::getEnvironment()`
  * `yarp::conf::environment::setEnvironment()`
  * `yarp::conf::environment::unsetEnvironment()`

#### `os`

* The method `yarp::os::getenv()` is [... cut]