robotology/yarp #1854: Migrate imuBosch_BNO055 driver to use the new YARP Multiple Analog Sensor Interfaces

traversaro, 2018-09-12:
See http://www.yarp.it/group__dev__iface__multiple__analog.html and https://github.com/robotology/yarp/issues/1526 .

See https://github.com/robotology/yarp/commit/43e162579d7d9971970b26f415250ac64afb4c00#diff-aaed599981554b9b315bdb83680e000dL41 for an example of a migration of a IMU-like device to the new interfaces while keeping compatibility with the old `IGenericSensor` interface.

traversaro, 2019-04-24:
cc @prashanthr05

prashanthr05, 2019-04-24:
This was done in the PR [!1910](https://github.com/robotology/yarp/pull/1910)

traversaro, 2019-04-24:
As for other issue listed in https://github.com/robotology/yarp/labels/Fixed%20in%3A%20YARP%20devel , I think we can close this issue as the problem was fixed in devel and will be released as part of YARP 3.2 .

closed 2019-04-24 by traversaro