robotology/yarp #1855: Make getDirectorySeparator aligned to std::filesystem::path::preferred_separator

Nicogene, 2018-09-12:
C++17 will introduce the `filesystem` header that has the `preferred_separator` constexpr defined as follow:

```c++
#ifdef _GLIBCXX_FILESYSTEM_IS_WINDOWS
    typedef wchar_t				value_type;
    static constexpr value_type			preferred_separator = L'\\';
#else
    typedef char				value_type;
    static constexpr value_type			preferred_separator = '/';
#endif
```

It is premature to introduce it in yarp since we are supporting only c++11( and filesystem is marked `experimental`) but my suggestion is to define this constexpr in `YARP_OS`, something like:

```c++
namespace yarp {
namespace os {
namespace filesystem {
#ifdef _GLIBCXX_FILESYSTEM_IS_WINDOWS
    typedef wchar_t				value_type;
    static constexpr value_type			preferred_separator = L'\\';
#else
    typedef char				value_type;
    static constexpr value_type			preferred_separator = '/';
#endif
} // namespace filesystem
} // namespace os
} // namespace yarp
```
In order to simply make it a typedef of `std::filesystem::path::preferred_separator` once we will support c++17.

Moreover I'd suggest to change `yarp::os::NetworkBase::getDirectorySeparator()` in:
```c++
std::string NetworkBase::getDirectorySeparator() {
    return yarp::os::filesystem::preferred_separator;
}
```

Deprecating it with a deprecation warning like (`"Use yarp::os::filesystem::preferred_separator instead"`).

I know that deprecating could be annoying for users, but I think that this function is used inside yarp but not so much used in user code.
Finally note that a `constexpr` is defined a compile time, then more optimized a function call.
Related to #1755 and #1853 .

traversaro, 2018-09-12:
:+1: , just pay attention that in YARP code the macro used should not be `_GLIBCXX_FILESYSTEM_IS_WINDOWS` , as this will not work if `libcxx` or the MSVC standard library is used.

traversaro, 2018-09-12:
By the way, I do not know the meaning of `L'\\'` ? Is that something C++17 or `libstdcxx`  specific?

Nicogene, 2018-09-12:
> By the way, I do not know the meaning of L'\\' ? Is that something C++17 or libstdcxx specific?
No idea, I found only this: https://stackoverflow.com/questions/6384118/what-does-the-l-in-front-a-string-mean-in-c

drdanz, 2018-09-12:
> By the way, I do not know the meaning of `L'\\'` ? Is that something C++17 or `libstdcxx` specific?

That's how you define a "wide char" `wchar_t` (standard c++98).

drdanz, 2018-09-12:
See https://en.cppreference.com/w/cpp/language/string_literal

Nicogene, 2018-09-13:
It is a "wide char" because `'\\'` has the dimension of 2 char am I right?

drdanz, 2018-09-14:
No, `\\` is one character represented in one byte (`char`), `L'\\'` is still a single character, but represented with 2 bytes (`wchar_t`)

drdanz, 2019-12-03:
Fixed by  #2062

PR #2062 (2019-12-03): Add filesystem utilities
It fixes #1855 .

The main feature brought by this PR is that it has been possible to remove the inclusion of `Network.h` in several places where the actual network or communication wasn't involved.

It is a draft since @drdanz has some plan to create a separate library and get rid of other stuff not concerning the network in `Network.h`
commit ad196162 (2019-12-03): conf: add yarp::conf::filesystem utilities

It fixes #1855.
closed 2019-12-03 by drdanz