robotology/yarp #1861: Wrong implementation of realsense2Driver::setParams()

claudiofantacci, 2018-09-18:
**Describe the bug**
`realsense2Driver::setParams()` returns `false` if parameter does not exist on camera and the driver is not run.

**To Reproduce**
Use Intel SR300 camera and run the driver. `RS2_OPTION_EMITTER_ENABLED` does not exist and the driver is not run.

**Expected behavior**
The device driver should write a warning that the parameter does not exist on the camera and the driver is run.

**Configuration (please complete the following information):**
Tested on
 - OS: Ubuntu, macOS
 - yarp version: `3.1.100+20180917.4+git1fcacde6c`
-  compiler: clang, gcc

**Additional context**
Here is the function that needs to be updated.

https://github.com/robotology/yarp/blob/2d160a4abb158e570d115d7774b29d06617bf981/src/devices/realsense2/realsense2Driver.cpp#L560-L668

With SR300, commenting line 612 fixes the problem and the device driver is run.
Probably is sufficient to change the behavior of the function by not using `realsense2::settingErrorMsg()` function.

Nicogene, 2018-09-18:
Instead of commenting out the line 612, have you tried to remove 
```
enableEmitter   true
```
from ini files? That error should be triggered by the fact you are saying to the device to set something it doesn't have.

If it doesn't work also removing that line from the ini file yes it is a bug, otherwise it is the expected behavior. 

Unfortunately I have not a device to test it

claudiofantacci, 2018-09-18:
I set the option to false, I did not delete the line.
If that`s the expected behavior then it`s fine, but my two cents here is that if the option is not available, the device should start anyway. Why wouldn`t you do anyway?
As a matter of the fact, If I set a feature/option to off, why would you not allow the device to start?
Also, this implies that if you have different realsense with different options, you have to ship a config file for each possible camera while you could just ship a single file ingoring options that are not allowd on a camera.
Am I missing something?

Nicogene, 2018-09-18:
If you put it in the config file to `false` the driver tries to contact the hardware to turn off the emitter that by default is on when the realsense is started.

If the specific device(in your case the SR300) doesn't have the emitter the call to the `realsense` SDK fails and this error is propagated to `setParam`.

For `depthCamera` and `realsense2` devices the parsing of the file has been designed that everything that it is under the group `[SETTINGS]` make to fail the open the device if that setting requested is not available. (cc @barbalberto and @aerydna  

Unfortunately this enforce you to know what kind of settings you can set on your specific device, so for each type of realsense camera you should have different config files, because actually they are different cameras, that share the same SDK.

If removing the option from the ini file doesn't make it fail I would not mark it as bug but a design choice. 
On the other hand, if it fails also if you remove the setting from the ini **it is** a bug :bug:

claudiofantacci, 2018-09-18:
> For `depthCamera` and `realsense2` devices the parsing of the file has been designed that everything that it is under the group `[SETTINGS]` make to fail the open the device if that setting requested is not available. (cc @barbalberto and @aerydna)
Unfortunately this enforce you to know what kind of settings you can set on your specific device, so for each type of realsense camera you should have different config files, because actually they are different cameras, that share the same SDK.

I changed the name of the issue because what we are discussing is the expected behavior and not a bug.

However, I feel that this is very limiting and somewhat counterintuitive for a device.
Basically the driver does not open because a requested feature is not present and it does not have an impact on the device itself.
Maybe this is ok to some particular devices where you need to have strong control on features and parameters, but I reckon this is very limiting on cameras.

Is there any room to discuss this 😄? Otherwise, good to know and we can close the issue.

claudiofantacci, 2018-09-18:
@barbalberto thanks for the comments and the explanation.
It is clear now the intents behind the behavior. Good to know 👍 
I'm closing the issue 😃

closed 2018-09-18 by claudiofantacci