robotology/yarp #1862: yarp conf and yarp namespace create folder instead of files

diegoferigo, 2018-09-18:
**Describe the bug**
On Windows both `yarp conf ip port` and `yarp namespace /namespacename` create folders instead of files. Configuring the environment is not anymore possible.

For what concern namespace a workaround is using the `YARP_NAMESPACE environment variable`.

**To Reproduce**
Execute `yarp conf ip port` or `yarp namespace /namespacename` of Windows.

Reverting to 2aa5e018e8ed8751fbca8b5ce9f97e5b97c05876 works fine.

**Expected behavior**
The ability of configuring a clean system.

**Configuration (please complete the following information):**
 - OS: Windows
 - yarp version: last master 2d160a4abb158e570d115d7774b29d06617bf981
-  compiler: Visual Studio 2017

Probably related to https://github.com/robotology/yarp/pull/1852 or https://github.com/robotology/yarp/pull/1841

cc @claudia-lat @mebbaid 
cc @drdanz @Nicogene @claudiofantacci

Nicogene, 2018-09-18:
It is something that we experienced also on Linux with @claudiofantacci .
@drdanz did the fix here  1ae11f9, are you sure that everything is installed correctly?

Nicogene, 2018-09-18:
BTW it is very strange that reverting that commit fixes the problem, it is in the `LocalBroker` so yarpmanager-related.

Nicogene, 2018-11-23:
Are you still experiencing this issue?
Otherwise I will close the issue

diegoferigo, 2018-11-23:
The machine where we faced this issue was updated some weeks ago and everything seems to work fine. Thanks!

closed 2018-11-23 by diegoferigo