robotology/yarp #1866: Unable to open devices that are dynamically compiled in windows

mebbaid, 2018-09-19:
**Describe the bug**
When doing `yarpdev --list` the following is displayed

[image]

When verbose with `yarp plugin --search-path,` the following message appears
[image]

**To Reproduce**
Update the robotology superbuild, and then update and build yarp

**Configuration (please complete the following information):**
 - OS: Windows 10
 - yarp version: latest master
-  compiler: VS

@traversaro

traversaro, 2018-09-19:
I think the output of `yarp plugin --search-path` is the most interesting thing. 

On my Linux machine the output is:
~~~bash
straversaro@iiticublap103:~$ yarp plugin --search-path
Search path:
             yarp:	/home/straversaro/src/robotology-superbuild/build/install/lib/yarp
             iCub:	/home/straversaro/src/robotology-superbuild/build/install/lib/iCub
      ICUBcontrib:	/home/straversaro/src/robotology-superbuild/build/install/lib/ICUBcontrib
           codyco:	/home/straversaro/src/robotology-superbuild/build/install/lib/codyco
~~~

I think this could be related to https://github.com/robotology/yarp/pull/1841 or https://github.com/robotology/yarp/issues/1513 . Can you try to see if the issue is present also in the commit 4d82b1e70238d00d45b5504d2e393c36089564dd (the one before  https://github.com/robotology/yarp/pull/1841)?

traversaro, 2018-09-19:
It would be  interesting also to understand why the `hde` search path is working fine. 
Can you paste the content of the `PATH` and `YARP_DATA_DIRS` env variables?

traversaro, 2018-09-19:
What is that %_DIR%` in the `PATH` ?

mebbaid, 2018-09-19:
Type while opening the variable, deleted it but nothing changed.

traversaro, 2018-09-19:
It seems that there was another change in the setup: you installed https://github.com/robotology-playground/wearables I guess on branch `feature/cleanup` and that repo installed the plugins in the same location of YARP, and I guess this is the reason why everything is working.

traversaro, 2018-09-19:
Just for reference, could you print the content of the following files: 
*  robotology-superbuild/build/install/share/yarp/plugins/yarp.ini 
*  robotology-superbuild/build/install/share/iCub/plugins/iCub.ini 
* robotology-superbuild/build/install/share/ICUBcontrib/plugins/ICUBcontrib.ini 
* robotology-superbuild/build/install/share/yarp/plugins/wearables.ini 

On my Linux machine they are: 
~~~
straversaro@iiticublap103:~/src$ cat
###### This file is automatically generated by CMake.
[search yarp]
path "/home/straversaro/src/robotology-superbuild/build/install/lib/yarp"
extension ".so"
type "shared"
straversaro@iiticublap103:~/src$ cat robotology-superbuild/build/install/share/iCub/plugins/iCub.ini 
###### This file is automatically generated by CMake.
[search iCub]
path "/home/straversaro/src/robotology-superbuild/build/install/lib/iCub"
extension ".so"
type "shared"
straversaro@iiticublap103:~/src$ cat robotology-superbuild/build/install/share/ICUBcontrib/plugins/ICUBcontrib.ini 
###### This file is automatically generated by CMake.
[search ICUBcontrib]
path "/home/straversaro/src/robotology-superbuild/build/install/lib/ICUBcontrib"
extension ".so"
type "shared"
~~~

mebbaid, 2018-09-19:
First file: yarp.ini
```
###### This file is automatically generated by CMake.
[search yarp]
path "C:\Users\icub\Documents\code\git\robotology-superbuild\build\install\lib\yarp"
extension ".dll"
type "shared"
```
Second file: iCub.ini
```
###### This file is automatically generated by CMake.
[search iCub]
path "C:\Users\icub\Documents\code\git\robotology-superbuild\build\install\lib\iCub"
extension ".dll"
type "shared"
```
Third File: ICUBcontrib.ini
```
###### This file is automatically generated by CMake.
[search ICUBcontrib]
path "C:\Users\icub\Documents\code\git\robotology-superbuild\build\install\lib\ICUBcontrib"
extension ".dll"
type "shared"
```
wearables.ini
```
###### This file is automatically generated by CMake.
[search wearable]
path "C:\Users\icub\Documents\code\git\robotology-superbuild\build\install\lib\yarp"
extension ".dll"
type "shared"
```

mebbaid, 2018-09-19:
Maybe this is why it finds all the plugins except those of icub, since the correct folder is like in the picture:
[image]

https://github.com/robotology/yarp/issues/1866#issuecomment-422745458

traversaro, 2018-09-19:
Cool, nice catch.

traversaro, 2018-09-19:
I checked on my system, and there should be both `ICUB` (for CMake config files) and `iCub` (for YARP plugin). Are you sure that the iCub project is actually installing those plugins?

mebbaid, 2018-09-19:
yes, however both the cmakeconfig and the plugins are in the same `ICUB` as in:
[image]

mebbaid, 2018-09-19:
Indeed editing the plugin .ini files path variables manually fixes the corresponding search path returned by `yarp plugin --search-path`
[image]

mebbaid, 2018-09-19:
`yarp.ini` after modifying manually:
```
###### This file is automatically generated by CMake.
[search yarp]
path "C:/Users/icub/Documents/code/git/robotology-superbuild/build/install/lib/yarp"
extension ".dll"
type "shared"
```

Yeshasvitvs, 2018-10-12:
@traversaro I ran into this issue today while trying to access the ftShoes. It seems like this problem is fixed in https://github.com/robotology/yarp/pull/1841 but the changes are not reflected yet. Any suggestions on how to proceed to fix it?

traversaro, 2018-10-12:
Hi @Yeshasvitvs , can you provide the output of your `yarpdev --list` and the content of the following files (before/after) your local fixes:

* robotology-superbuild/build/install/share/yarp/plugins/yarp.ini
* robotology-superbuild/build/install/share/iCub/plugins/iCub.ini
* robotology-superbuild/build/install/share/ICUBcontrib/plugins/ICUBcontrib.ini
*  robotology-superbuild/build/install/share/yarp/plugins/wearables.ini

Yeshasvitvs, 2018-10-12:
@traversaro I am using the following YARP master branch with `HEAD` at 

```
commit d1a2effb61bd8888e7b19701d2b6bd24e4f30dd4 (HEAD -> master, origin/master, origin/HEAD)
Merge: 1c95e98c8 4624e7c1e
Author: Daniele E. Domenichelli <daniele.domenichelli@gmail.com>
Date:   Thu Oct 11 11:08:21 2018 +0200

    Merge pull request #1891 from drdanz/fix_VAS_status

    Fix -Wattributes warning for yarp::dev::VAS_status
```

### Before local fixes

[image]

### After local fixes

[image]

**NOTE:** I am not using `wearables` yet

traversaro, 2018-10-12:
Ok, that seems to be super clear. I am surprised because I would expect @claudiofantacci , @pattacini or other regular Windows users to be affected as well.

traversaro, 2018-11-02:
We experienced the same issue with @AngeloRendiniello with latest YARP master branch.

traversaro, 2018-11-02:
I think I understood what is going on: @pattacini and (I guess) @claudiofantacci use devices from the `build` tree, not the `install` tree, and indeed if you check in: 
https://github.com/robotology/yarp/pull/1841/files#diff-cd0eb3dc7250fb8863489a632361de81R92 
the directory separator are not changed in the `_path` of the `yarp.ini` installed in the build `tree`, but it is just modified for the installed one. Fix provided in https://github.com/robotology/yarp/pull/1911 .

closed 2018-11-05 by drdanz