robotology/yarp #1868: Should IFrameGrabber and IFrameGrabberRgb be deprecated?

traversaro, 2018-09-21:
The `IFrameGrabber` and the `IFrameGrabberRgb` interfaces seem to have been superseded by the `IFrameGrabberImage` and the `IFrameGrabberImageRaw` interfaces since a long time, they are not used in any network driver (not even for back-compatibility) and can be found only in some devices in  `icub-main` ( https://github.com/robotology/icub-main/search?q=IFrameGrabberRgb&unscoped_q=IFrameGrabberRgb), but all those devices  also offer a variant that implements the modern interfaces `IFrameGrabberImage` and the `IFrameGrabberImageRaw` . 

I am not an expert of this part of YARP, but I guess this interfaces can be safely deprecated and eventually removed from YARP.

@vtikha @barbalberto @drdanz @Nicogene

Nicogene, 2018-09-24:
I found out that `IFrameGrabberRgb` and `IFrameGrabber` are used in `dragonfly2` and `usbCamera`.

@barbalberto , are they easily portables to `IFrameGrabberImage` and `IFrameGrabberImageRaw`?

If so we can deprecate the old ones as @traversaro suggested.

traversaro, 2018-09-24:
`dragonfly2` and `usbCamera` actually come two versions, one using the old interfaces and another using the new one. It is a bit tricky to understand this because both devices are in the same source code file.

barbalberto, 2018-09-24:
In my opinion they can be deprecated.
The only update required is in the [usbCamera](https://github.com/robotology/icub-main/blob/c3ad642ed0aacdb3216f692cc73916936079714b/src/libraries/icubmod/usbCamera/common/USBcamera.cpp#L382) where the new interface is calling the old one of the lower level implementation.

It should propagate the new one and the low level device shall implement it [here ](https://github.com/robotology/icub-main/blob/c3ad642ed0aacdb3216f692cc73916936079714b/src/libraries/icubmod/usbCamera/linux/V4L_camera.cpp#L922)

Few changes are required, not a big deal.

drdanz, 2018-10-04:
I made a quick test commenting out these interfaces, and I can confirm that they are completely unused in YARP.

drdanz, 2019-04-11:
usbCamera is now in YARP, therefore we should update it before removing the interface

closed 2020-12-22 by drdanz