robotology/yarp #1870: Refactor image horizontal concatenation logic

traversaro, 2018-09-21:
**Is your feature request related to a problem? Please describe.**
Combining two images as one single image in which the images are horizontally concatenated is  a technique that is frequently used in YARP, for example to transmit two images in a synchronous way, as supported for example by the `grabberDual` device. However,  the code for horizontally concatenate is currently duplicated in two points of YARP, i.e. https://github.com/robotology/yarp/blob/f5e903f58cd4550504d6f6643e0bb3f91ece2ffe/src/libYARP_dev/src/devices/ServerGrabber/ServerGrabber.cpp#L1001 and https://github.com/robotology/yarp/blob/3d550be4ea1f51f7903a88f2279eaa85948f44df/src/devices/realsense2/realsense2Driver.cpp#L1495 . Given that I will copy once more time now, so [I guess](https://en.wikipedia.org/wiki/Rule_of_three_%28computer_programming%29) we should refactor it as a free function. 
Any opinion on the suitable header that should host this function?  

**Describe the solution you'd like**
Have that  a code in reusable function. 

**Describe alternatives you've considered**
Copy-paste the code.

claudiofantacci, 2018-09-23:
I totally agree 😄 It can be very useful also for visualization purposes 👍 

cc @barbalberto @Nicogene

Nicogene, 2018-09-24:
I agree, since it has not particular dependencies this code could be moved in `YARP_sig` as a sort of extension of `Image` functionalities.
Good catch, code duplication is evil :imp:

Nicogene, 2018-10-03:
Fixed by #1880

PR #1880 (2018-10-01): Add ImageUtils
This PR add  `ImageUtils.h`, an header containing the following image utilities:
  - `utils::vertSplit`
  - `utils::horzSplit`
  - `utils::vertConcat`
  - `utils::horzConcat`

it fixes #1870 

Please review code
commit a9d3fa69 (2018-10-01): YARP_sig: add ImageUtils, an header containing image utilities.

Utilities provided:
- `vertSplit`
- `horzSplit`
- `horzConcat`
- `vertConcat`
It fixes #1870
closed 2018-10-03 by Nicogene