robotology/yarp #1875: RGBDSensorWrapper example fail to run

traversaro, 2018-09-25:
**Describe the bug**
The command described in https://github.com/robotology/yarp/blame/b7983c941952a3aafa7de7dd4aeeb574c1865799/example/dev/RGBD/README.md#L21 : 
~~~
yarpdev --device RGBDSensorWrapper --name /server --subdevice fakeDepthCamera --mode ball 
~~~
is failing with the following error:
~~~
straversaro@iiticublap103:~/src/idjl-superbuild/build3/idjl/YARP$ yarpdev --device RGBDSensorWrapper --name /server --subdevice fakeDepthCamera --mode ball 
[INFO]RGBDSensorWrapper: using default 'period' parameter of  0.03 s 
[INFO]RGBDSensorWrapper: ROS configuration parameters are not set, skipping ROS topic initialization. 
yarp: Port /server/rpc:i active at tcp://192.168.1.10:10007/
yarp: Port /server/rgbImage:o active at tcp://192.168.1.10:10008/
yarp: Port /server/depthImage:o active at tcp://192.168.1.10:10009/
[DEBUG]opening IRGBDSensor subdevice
[INFO]Test grabber period 0.0333333 / freq 30 , mode [ball]
[INFO]created device <test_grabber>. See C++ class yarp::dev::TestFrameGrabber for documentation.
[INFO]created device <fakeDepthCamera>. See C++ class yarp::dev::fakeDepthCameraDriver for documentation.
[ERROR]RGBD wrapper: error configuring interfaces for parsers 
[ERROR]RGBDSensorWrapper: error while opening subdevice
yarpdev: ***ERROR*** driver <RGBDSensorWrapper> was found but could not open
[ERROR]yarpdev: ***ERROR*** device not available.
[INFO]Suggestions:
[INFO]+ Do "yarpdev --list" to see list of supported devices.
[INFO]+ Or append "--verbose" option to get more information.
~~~

**Expected behavior**
The one described in the tutorial. 

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04
 - yarp version: current `devel` 
-  compiler: Gcc 7.3

PeterBowman, 2018-09-25:
`fakeDepthCamera` needs to implement `yarp::dev::FrameGrabberControls`. `yarp::dev::RGBDSensorWrapper` expects this interface in its subdevices since YARP 2.3.72.1.

traversaro, 2018-09-25:
Thanks @PeterBowman .

PR #1878 (2018-09-26): Various fixes rgbd sensors/wrapper
This PR:
- Fixes #1871
- Fixes #1875
- Fixes #1877 

Moreover it clarifies the documentation of (`set/getDepthAccuracy`).

Tests on real and fake device have been done.

Please review code.
commit ad654953 (2018-09-26): RGBDSensorWrapper: make the view of IFrameGrabberControl optional.

It fixes #1875
closed 2018-09-26 by Nicogene