robotology/yarp #1883: YARP CMake package is not relocatable

InesSorrentino, 2018-10-02:
Running find_package(YARP …) with the Windows binaries result in the following error:
~~~
CMake Error at C:/Program Files/robotology/yarp-3.0.1/cmake/YARP_idl_toolsConfig.cmake:22 (include):
  include could not find load file:

    E:/Cygwin64/home/icub/yarp-packaging/windows/build/yarp/cmake/YarpIDL.cmake
Call Stack (most recent call first):
  C:/Program Files/robotology/yarp-3.0.1/cmake/YARPConfig.cmake:183 (find_package)
  C:/Program Files/robotology/ycm-0.9/share/YCM/modules/FindOrBuildPackage.cmake:180 (find_package)
  CMakeLists.txt:109 (find_or_build_package)

CMake Error at C:/Program Files/robotology/yarp-3.0.1/cmake/YARP_idl_toolsConfig.cmake:22 (include):
  include could not find load file:

    E:/Cygwin64/home/icub/yarp-packaging/windows/build/yarp/cmake/YarpIDL.cmake
Call Stack (most recent call first):
  C:/Program Files/robotology/yarp-3.0.1/cmake/YARPConfig.cmake:183 (find_package)
  C:/Program Files/robotology/ycm-0.9/share/YCM/modules/FindOrBuildPackage.cmake:287 (find_package)
  CMakeLists.txt:109 (find_or_build_package)
~~~

PeterBowman, 2018-10-02:
This is a known issue I reported shortly after the YARP 3.0.1/3.1.0
release. The `yarp-3.0` branch includes a hotfix that couldn't be
backported in time for the 3.0.x release cycle. It is therefore fixed in
current master.

PeterBowman, 2018-10-02:
For the record, see https://github.com/robotology/QA/issues/313 and https://github.com/robotology/yarp/commit/479d0755d64443c869454c295c9b1cdd311f031a.