robotology/yarp #1885: Set EXTRACT_PRIVATE to FALSE avoid the use of DOXYGEN_SHOULD_SKIP_THIS

traversaro, 2018-10-03:
YARP is full of `DOXYGEN_SHOULD_SKIP_THIS` preprocessor checks (see https://github.com/robotology/yarp/search?q=DOXYGEN_SHOULD_SKIP_THIS&unscoped_q=DOXYGEN_SHOULD_SKIP_THIS ). Most of them seem to be used to hide private methods and attributes. 
If our objective is to hide private methods and attributes, perhaps it is easier to just set the [`EXTRACT_PRIVATE` Doxygen](https://www.stack.nl/~dimitri/doxygen/manual/config.html#cfg_extract_private) option in https://github.com/robotology/yarp/blob/1964482a5aeecad187af6ba45d31c14eb1ea0d77/cmake/template/Doxyfile.in#L452 to `FALSE` . 

cc @jeljaik

Nicogene, 2018-10-03:
Or even we can remove it, the default is `NO`

drdanz, 2018-10-10:
I wondered several times if the flag was actually useful to someone, and why `EXTRACT_PRIVATE` was enabled, see e9b8d16deab09b19a5f8ec9974903c117f6d4ac8 [*document more of PortCore operations*].
I didn't have the courage to change it, though, I was afraid that it could actually include some important documentation (same reason why I never disabled the documentation for the *impl* classes).

I'm in favour to remove it, but I'd like to hear some other comments by @lornat75 and by people actually using the documentation...