robotology/yarp #1888: Specified assumed motor serialization in YARP motor interfaces

traversaro, 2018-10-08:
**Is your feature request related to a problem? Please describe.**
For motor-related YARP interfaces (such as [`IMotorEncoder`](https://github.com/robotology/yarp/blob/f5e903f58cd4550504d6f6643e0bb3f91ece2ffe/src/libYARP_dev/include/yarp/dev/IMotorEncoders.h) ) there is no way to know to which motor the measured quantities are relative to. This implicit information is quite obvious for robots with direct coupling (one motor always actuates one joint) but it is a common source of errors for coupled joints (such as the iCub 1/2's torso and shoulder). 

**Describe the solution you'd like**
Possible solutions:
* Add a `getMotorName` method to the [`IAxisInfo`](http://www.yarp.it/classyarp_1_1dev_1_1IAxisInfo.html) interface
* Add a new `IMotorInfo` interface, that contains the `getMotorName` method? This IMotorInfo could also provide coupling information (the [OPC UA Companion Specification for Robotics](https://opcconnect.opcfoundation.org/2018/06/opc-ua-machine-vision-and-robotics-specifications-released/) contains an information model for coupling, we could copy its semantics to avoid engineering a custom solution).

randaz81, 2018-10-10:
uhm, in the first case (IAxisInfo), the interface works on a joint basis. What is the return value of `getMotorName` that you recommend in the case of a coupled joint? An array of ids?

traversaro, 2018-10-10:
> uhm, in the first case (IAxisInfo), the interface works on a joint basis. What is the return value of getMotorName that you recommend in the case of a coupled joint? An array of ids?

I was thinking of having some methods working on a joint basis (`getAxisName`/`getJointType`) and another  working on a motor-basis, something like `bool getMotorName (int motor, std::string &motorName)=0` . Perhaps this is too confusing, and that would be a point in favor of a separate `IMotorInfo`.

randaz81, 2018-10-10:
To quickly recap our f2f meeting:

- it's ok to add `getMotorName (int motor, std::string &motorName)` in a new interface `IMotorInfo `👍 
- if we need to get the list of motors associated to a particular joint, this should not be in `IMotorInfo`, since the input is a joint and not a motor, and it would be nice to avoid promiscuous interfaces. i.e. all the methods belonging to an interface should have, if possible, consistent inputs (joints ids OR motors ids). So this method should belong to IAxiaInfo (or another interface).