robotology/yarp #1901: Methods to check the library version at run time are missing

drdanz, 2018-10-18:
**Is your feature request related to a problem? Please describe.**
There is no way to check the version of the library loaded at runtime.

As described in https://github.com/robotology/QA/issues/325, there is a way to check the version of the library at build time, but no way to know the version of the library loaded at run time.

**Describe the solution you'd like**
We should add some methods `yarp::conf::getVersion()`, `yarp::conf::getMajorVersion()`, etc.

Nicogene, 2018-10-19:
And maybe change 
```
std::string Companion::version() {
    return YARP_VERSION;
}
```
in order to call `yarp::conf::getVersion()`.

traversaro, 2018-10-19:
If that method is not inline in the header it should do what @drdanz intended.

drdanz, 2018-10-19:
> If that method is not inline in the header it should do what @drdanz intended.

Exactly, the only issue is that if we move it in `yarp::conf` we need to have an actual library (at the moment it's header only).

drdanz, 2018-10-20:
Actually I realized that having `yarp::conf::getVersion()` makes no sense, it would make more sense to have `yarp::os::getVersion()`, `yarp::sig::getVersion()`, etc...