robotology/yarp #1918: Could not save configuration file /home/rainalex/.config/yarp/yarp.conf - error when running yarpserver

RainAlexandra, 2018-11-10:
When I execute the `yarpserver` command (I also tried running `yarp server` but this did not change anything) the configuration file is not saved. So I get this error:

```
Could not save configuration file /home/rainalex/.config/yarp/yarp.conf
warning: YARP name server(s) not configured, ports will be anonymous
warning: check your namespace and settings with 'yarp detect'
yarp: Port /root active at tcp://100.88.4.107:10000/
```

Even if I run `yarp detect` while the yarp name server is running all the tips given do not help at all because

1. the yarp name server is indeed running
2. the yarp namespace is set as /root
3. `yarp conf` returns the path to the configuration file (`/home/rainalex/.config/yarp/yarp.conf`) which does not actually exist 

Am I not running the correct commands? I've just been trying to check my installation by following the instructions that were given on [this iCub wiki page](http://wiki.icub.org/wiki/Check_your_installation)

drdanz, 2018-11-10:
Which YARP version are you using?

traversaro, 2018-11-10:
Are you sure that the user that is executing the `yarpserver` have write access to the `.config/yarp` directory? Which operating system are you using?

closed 2018-11-12 by RainAlexandra