robotology/yarp #1921: PontCloudTypes: Possible error in DataRGBA

drdanz, 2018-11-19:
The order of the bytes in `DataRGBA` seems wrong (`b`, `g`, `r`, `a`).

https://github.com/robotology/yarp/blob/ff713f369ff3a00aafd64d8105597e203f58b65a/src/libYARP_sig/include/yarp/sig/PointCloudTypes.h#L483-L490

Is this a typo? If it is not, please add a comment to explain...

drdanz, 2018-11-19:
CC @Nicogene

Nicogene, 2018-11-19:
The point cloud types have been taken from PCL, see [their documentation](http://docs.pointclouds.org/trunk/point__types_8hpp_source.html) at line 224.
Sincerely I have no idea if they did with some rationale or it is a typo

claudiofantacci, 2018-11-20:
Interesting observation. PCL defines
[PCL_ADD_UNION_RGB](https://github.com/PointCloudLibrary/pcl/blob/c26faa2a696432cb629b7fe012de03f71db0762a/common/include/pcl/impl/point_types.hpp#L224-L239) as a **BGR** union, then it defines [RGB vectors](https://github.com/PointCloudLibrary/pcl/blob/c26faa2a696432cb629b7fe012de03f71db0762a/common/include/pcl/impl/point_types.hpp#L242-L247) reordering the data in the union. I guess that `PCL_ADD_UNION_RGB` could be simply renamed `PCL_ADD_UNION_BGR`. As long as there is an agreement on data alignment, our structure is good to go, isn't it?

traversaro, 2018-11-20:
By the way, I found a lot of this data structures in OpenCV, PCL, ROS, but I have never found a reference describing in detailed all of this.

Nicogene, 2018-11-20:
>  As long as there is an agreement on data alignment, our structure is good to go, isn't it?

The problem that I see here is that if I access to `r, g, b, a` through the `NetInt32 rgba`,  I could think that its first byte is the `r` and it is the `b` instead. Am I right?
Obviously I don't think that so many people use this struct as `NetInt32` but it could cause issues

claudiofantacci, 2018-11-20:
You are definitely right. I'm not sure what would be the proper way to tackle this. PCL defines RGB types as BGR, providing proper documentation [here](http://docs.pointclouds.org/trunk/structpcl_1_1_point_x_y_z_r_g_b.html).
If we are using that strucutre to be a replica of PCL, then I guess we need to leave the structure as it is.

drdanz, 2018-11-20:
It doesn't look like bgr to me, r is shifted by 16 and g is shifted by 8...

drdanz, 2018-11-20:
Actually I was wrong, being on little endian system, this is `B` `G` `R` `_`.

barbalberto, 2018-11-20:
> Actually I was wrong, being on little endian system, this is B G R _.

Yes, I think the definition leaverage on the little endiannes... at this point the shouldn't the int be *Argb*?
Otherwise I maybe tempted to do:
` rgba = (r) << 24 | (g) << 16 | (b) <<8 | (a) ; `
which is incorrect, right?

drdanz, 2018-11-20:
Either `BGRA` or `ARGB`, but for sure not `RGBA`

drdanz, 2019-04-11:
Closing this. The data is correct (the struct comes from PCL), the name is confusing.

closed 2019-04-11 by drdanz