robotology/yarp #1925: AnalogWrapper multiple ros topics for WrenchStamped ros msgs

Yeshasvitvs, 2018-11-28:
**Is your feature request related to a problem? Please describe.**
Currently, [AnalogWrapper.cpp](https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/devices/AnalogWrapper/AnalogWrapper.cpp) opens only a single ros port and in the case of using `geometry_msgs/WrenchStamped` ros msg type only the first 6 elements are read from the attached device as this logic is [hardcoded](https://github.com/robotology/yarp/blob/e1d7a45621ea97a76136f94511a3b4c956506bf5/src/libYARP_dev/src/devices/AnalogWrapper/AnalogWrapper.cpp#L811).  

**Describe the solution you'd like**
Ideally I would like to have the functionality to open the same number of ros topics as that of the number of `ports` used in the config file. In this way each rostopic handles the data corresponding to those mentioned in the `ports`. This will enable to use a single `analogWrapper` to handle multiple data from the attached subdevice.

traversaro, 2018-11-28:
This seems to be something for which it could make sense to use new IMultipleAnalogSensors interface: I think the ROS-specific logic of `AnalogWrapper` could be move to a ROS-enabled IMultipleAnalogSensors wrapper, that could publish an arbitrary number of sensors exposed by a device exposing IMultipleAnalogSensors, even of different types.

Yeshasvitvs, 2018-11-28:
@barbalberto @nico202 I added the feature and opened a [PR](https://github.com/robotology/yarp/pull/1926). I make sure that the current configuration is still supported by launching the icub model in gazebo and for the case of testing multiple ros topics, I used this [config file](https://github.com/Yeshasvitvs/human-dynamics-estimation/blob/c504533a7981c3d054717be426ce30bcc2f467ae/app/conf/newhde.xml#L239). The rostopics handle the data as needed and I visualized this data using Rviz. Please review this PR.

randaz81, 2023-09-04:
Closing because AnalogWrapper is now deprecated for ROS msgs