robotology/yarp #1927: Provide callback function to Port::setReader

jbbjarnason, 2018-11-30:
**Is your feature request related to a problem? Please describe.**
When a class has several readers, a secondary processor class is required for new state callback.

**Describe the solution you'd like**
Simplify creating new state callback:
`
    auto portIn = std::unique_ptr<yarp::os::Port>(new yarp::os::Port);
    portIn->open("/foo");
    portIn->setReader([this](yarp::os::ConnectionReader& connection){
        // do something
    });
`
also providing port name would simplify:
`
    auto portIn = std::unique_ptr<yarp::os::Port>(new yarp::os::Port("/foo"));
    portIn->setReader([this](yarp::os::ConnectionReader& connection){
        // do something
    });
`

**Describe alternatives you've considered**
Extending port class with another class and using the new class to create ports.

Nicogene, 2018-11-30:
Hi @jbbjarnason,

in the `yarp::os::Port` and `yarp::os::BufferedPort` the callback is handled in a old-fashioned way using inherithance, instead of using function pointers/lambdas.

In the current state is quite tricky, we considered to implement the callback mechanism in a more easy and modern way.
I think it would be more suitable to implement another method with a more self explanatory name like `setReadCallback(...)` and `setWriteCallback(..)` that accept function pointer, and keep `setReader(...)` as it is in order to not break the user/legacy code.

jbbjarnason, 2019-05-21:
Hi, I am using now zeromq instead now. Just to let know that I will not follow up on this issue. The issue may be closed.