robotology/yarp #1935: [Bindings, C#] Alias 'yarp' not found in generated InteractionModeEnum.cs / JointTypeEnum.cs / YarpVocabPixelTypesEnum.cs

kosenhitatchi, 2018-12-19:
Good afternoon,

I facing some issue in the .cs files generated by SWIG.

I'm on ubuntu 18.04 with yarp v:3.1.0 installed (from binary),  and  I'm compiling  the yarp C# binding with the information provided on the yarp documentation (build separately from YARP):
[www.yarp.it/yarp_swig.html#yarp_swig_configure](www.yarp.it/yarp_swig.html#yarp_swig_configure)

My goal is to be able to run this code ([Unity-Yarp-Integration](https://github.com/dcam0050/Unity-Yarp-Integration))  but the binding library need to be recompiled.
I already tried with the existing library in /Assets/Plugins/yarp_cs.so.

**Describe the bug**
With the code copied from [the yarp source](https://github.com/robotology/yarp/tree/55f066a4f3114c1a4995c106117078b07dda6bf6) (commit 55f066a4f3114c1a4995c106117078b07dda6bf6) I'm able to compile the C# binding.
Despite a lot of SWIG  warning (log available on demand), the compilation seems to went fine.

However when i open my IDE (monodevelop or vscode) I always get the same error on the files: _InteractionModeEnum.cs_ / _JointTypeEnum.cs_ /  _YarpVocabPixelTypesEnum.cs_

**_InteractionModeEnum.cs_** (idem for the two others file)

```
public enum InteractionModeEnum {
  VOCAB_IM_STIFF = yarp::os::createVocab('s','t','i','f'),
  VOCAB_IM_COMPLIANT = yarp::os::createVocab('c','o','m','p'),
  VOCAB_IM_UNKNOWN = yarp::os::createVocab('u','n','k','n')
}
```

> Alias  'yarp' not found [Assembly-CSharp-firstpass, Assembly-CSharp, Assembly-CSharp-Editor]

>Unexpected use of an aliased name [Assembly-CSharp-firstpass, Assembly-CSharp, Assembly-CSharp-Editor]

Before doing any modification in the _yarp.i_ i tried to fix this issue the dirty way by changing in the .cs file :
 ```VOCAB_IM_STIFF = yarp::os::createVocab('s','t','i','f') ``` to
`  VOCAB_IM_STIFF = yarp.createVocab('s','t','i','f')`
But i get the following error : 
> The expression being assigned to  'InteractionModeEnum.VOCAB_IM_STIFF' must be constant [Assembly-CSharp-firstpass]

Do you know how to fix this ? Or at least a known functional version of the C# binding ?

Many thanks

**Configuration (please complete the following information):**
 - OS: Ubuntu 18.04.1 LTS
 - yarp version: YARP version 3.1.0~dev
-  CMake Version 3.10.2
-  compiler:  (cf below)

**Additional context**

`dpkg --list | grep compiler`
```
ii  antlr                                                       2.7.7+dfsg-9.2                                              all          language tool for constructing recognizers, compilers etc
ii  fsharp                                                      4.1.33-0xamarin7+ubuntu1804b1                               all          functional-first programming language - compiler for the CLI
ii  g++                                                         4:7.3.0-3ubuntu2.1                                          amd64        GNU C++ compiler
ii  g++-7                                                       7.3.0-27ubuntu1~18.04                                       amd64        GNU C++ compiler
ii  gcc                                                         4:7.3.0-3ubuntu2.1                                          amd64        GNU C compiler
ii  gcc-7                                                       7.3.0-27ubuntu1~18.04                                       amd64        GNU C compiler
ii  gfortran                                                    4:7.3.0-3ubuntu2.1                                          amd64        GNU Fortran 95 compiler
ii  gfortran-7                                                  7.3.0-27ubuntu1~18.04                                       amd64        GNU Fortran compiler
ii  libantlr-java                                               2.7.7+dfsg-9.2                                              all          language tool for constructing recognizers, compilers etc (java library)
ii  libecj-java                                                 3.13.3-1                                                    all          Eclipse Java compiler (library)
ii  libllvm6.0:amd64                                            1:6.0-1ubuntu2                                              amd64        Modular compiler and toolchain technologies, runtime library
ii  libmono-compilerservices-symbolwriter4.0-cil                5.16.0.220-0xamarin4+ubuntu1804b1                           all          Mono.CompilerServices.SymbolWriter library (for CLI 4.0)
ii  libxkbcommon0:amd64                                         0.8.0-1ubuntu0.1                                            amd64        library interface to the XKB compiler - shared library
ii  mono-mcs                                                    5.16.0.220-0xamarin4+ubuntu1804b1                           all          Mono C# 2.0 / 3.0 / 4.0 / 5.0  compiler for CLI 2.0 / 4.0 / 4.5
ii  mono-roslyn                                                 5.16.0.220-0xamarin4+ubuntu1804b1                           all          Microsoft C# compiler
```

traversaro, 2018-12-19:
Which version of swig are you using? The system `3.0.12-1` one?

kosenhitatchi, 2018-12-19:
Thanks for your super quick feedback.

> Which version of swig are you using? The system `3.0.12-1` one?

`swig -version`
```
SWIG Version 3.0.12
Compiled with g++ [x86_64-pc-linux-gnu]
Configured options: +pcre
```

traversaro, 2018-12-19:
> I'm on ubuntu 18.04 with yarp v:3.1.0 installed (from binary), and I'm compiling the yarp C# binding with the information provided on the yarp documentation (build separately from YARP):
www.yarp.it/yarp_swig.html#yarp_swig_configure

If I remember correctly, @jgvictores was an heavy user of the workflow "create and compile bindings for the binary-installed version of YARP", but in Windows in his case. 
How are you obtaining the `yarp.i` file and  the related CMakeLists.txt ? From the binary packages? If you are cloning the YARP repo, I suggest to try the `v3.1.0` ( https://github.com/robotology/yarp/tree/v3.1.0 ) to be compatible with your binary version.

kosenhitatchi, 2018-12-19:
@traversaro  I've installed yarp with the straightforward  package installation in Ubuntu.
As described here : http://www.yarp.it/install_yarp_linux.html

```
> sudo sh -c 'echo "deb http://www.icub.org/ubuntu bionic contrib/science" > /etc/apt/sources.list.d/icub.list'
> sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 57A5ACB6110576A6
> sudo apt-get update
> sudo apt-get install yarp
```

I'm getting the ` yarp.i ` and CMakeLists.txt in the binding folder form this repo (commit 55f066a4f3114c1a4995c106117078b07dda6bf6):

```
cd $YARP_ROOT/bindings
mkdir build
cd build
ccmake ..
> CREATE_CSHARP =ON
make
```
I'm going to try to redo the same procedure with the the `v3.1.0` ( https://github.com/robotology/yarp/tree/v3.1.0 ) and keep you updated.

jgvictores, 2018-12-19:
@traversaro Thanks for the ping! SWIG is interesting and useful! I've checked the state of what is installed by `yarp_3.1.0_v14_x86_amd64_1.exe` and it looks good, `no need to download anything else from yarp` philosophy, all (`CMakeLists.txt`, `yarp.i`, `*.i` and folders-per-language) nicely in `C:\Program Files\robotology\yarp-3.1.0\share\yarp\bindings`. I'll report if I find any issues on that side.

I do not think this is related to binary packages/installers. It looks like a `CSHARP (C#) + SWIG + VOCAB` issue. Relative comments of mine are:
- https://github.com/robotology/yarp/issues/1719#issuecomment-393945583: `C# but only compilation`
- https://github.com/robotology/yarp/issues/1719#issuecomment-393958909: `In theory, the 0 value is only enforced in C# (see 3, below). : )`

I'm really not sure how @nunoguedelha sorted it all out, but: `C# + SWIG + VOCAB` cannot be considered stable in

kosenhitatchi, 2018-12-19:
> I'm going to try to redo the same procedure with the the `v3.1.0` ( https://github.com/robotology/yarp/tree/v3.1.0 ) and keep you updated.

@traversaro I tried the same procedure with v3.1.0 ( https://github.com/robotology/yarp/tree/v3.1.0 ) , in my case I'm getting bunch of **error** in  yarpCSHARP_wrap.cxx.
With the current state of the repo (commit 55f066a4f3114c1a4995c106117078b07dda6bf6) I'm not getting any compilation issue.

@jgvictores my final target being windows, I'll try to reinstal yarp using `yarp_3.1.0_v14_x86_amd64_1.exe `

kosenhitatchi, 2018-12-19:
Thanks @traversaro for your reply.
As mentioned in #1719 the combination of `C# + SWIG + VOCAB` still seems under active development.

@traversaro as you seem to be a SWIG expert, do you know if there if there's a way (even a dirty one implying modifying the .cs) to obtain a working binding which do not involve VOCAB with my current distribution ( Ubuntu 18.04 + Swig 3.0.12)  ?

Thanks

traversaro, 2018-12-19:
> do you know if there if there's a way (even a dirty one implying modifying the .cs)

Can you check how other constants such as `VOCAB_CM_POSITION` are defined in the bindings code? You can probably copy how they are defined. The most extreme workaround is check to which integer those vocabs correspond (for example by printing those values as integer in C++) and you directly  `yarp::os::createVocab('s','t','i','f')` with the corresponding integer (but to be honest I do not know if you can do that with ENUMs in C#).

kosenhitatchi, 2018-12-19:
@traversaro I did not manage to find other *.cs file embeding VOCAB_CM_POSITION outside of the class `yarpPINVOKE.cs `, which get seems to get the method straight for the DLL.

However I fixed my problem (like a savage) by removing from my project :
_`InteractionModeEnum.cs`_ / _`JointTypeEnum.cs`_ /  _`YarpVocabPixelTypesEnum.cs`_

And commenting their call in:
_`IInteractionMode.cs`_ (l.57: _setInteractionMode()_ & l.72: _getInteractionMode()_) 
_`IInteractionModeRaw.cs`_ (l.57: _setInteractionModeRaw()_ )