robotology/yarp #1936: Unable to compile yarp_ovrheadset

GiulioRomualdi, 2018-12-19:
**Describe the bug**
When I try to compile `yarp_ovrheadset` I get the following error.
~~~
a reference of type "yarp::dev::iBattery::Battery_status&" (not const-qualified) cannot be initialized with a value of type "int"
~~~
**Screenshots**
[image]

**Configuration:**
 - OS: Windows10
 - yarp version: latest commit in devel https://github.com/robotology/yarp/commit/053d25c23164c7c4a46e056036c6bd17a60ed429
-  compiler: VisualStudio 2017

**Additional context**
The bug can be easily fixed by editing https://github.com/robotology/yarp/blob/053d25c23164c7c4a46e056036c6bd17a60ed429/src/devices/ovrheadset/TextureBattery.cpp#L148
into 
```cpp
yarp::dev::iBattery::Battery_status status;
```

cc @aerydna @kouroshD

drdanz, 2018-12-28:
@randaz81 I think this might be related to your recent changes to the `Battery` stuff

randaz81, 2019-01-04:
424d772 is fine, please proceed...

Nicogene, 2019-01-09:
Fixed by #1943 

Shall we cherry-pick the fix in `master` ?

closed 2019-01-09 by Nicogene

drdanz, 2019-01-10:
I think that the changes to the battery are in devel only

PR #1943 (2019-01-09): Fix yarp_ovrheadset build
This PR fixes #1936
commit 6008646b (2019-01-09): Fix #1936