robotology/yarp #1937: Improve the using of the TransformServer inside ovr_headset device

GiulioRomualdi, 2018-12-19:
I'm currently using (with @kouroshD  and @S-Dafarra) the `ovr_headset` device, and I noticed that device use a transformClient to publish the pose (position and orientation) of the left and right hands with respect the `"mobile_base_body_link"`. It would be useful to have also the homogeneous transformation of the frame attached to the head with respect the `"mobile_base_body_link"`.

Furthermore, I've noticed that one of the inputs of the device is the name of the root frame (see [here](https://github.com/robotology/yarp/blob/master/src/devices/ovrheadset/OVRHeadset.cpp#L439)) however the name of the root link seems to be hardcoded  [here](https://github.com/robotology/yarp/blob/master/src/devices/ovrheadset/OVRHeadset.cpp#L980-L987):
```cpp
    if (relative)
    {
     ....
    }
    else
    {
        OVR::Quatf lRot = OVR::Quatf(ts.HandPoses[ovrHand_Left].ThePose.Orientation)  * OVR::Quatf(OVR::Vector3f(0, 0, 1), M_PI_2);
        OVR::Quatf rRot = OVR::Quatf(ts.HandPoses[ovrHand_Right].ThePose.Orientation) * OVR::Quatf(OVR::Vector3f(0, 0, 1), M_PI_2);
        tfPublisher->setTransform(left_frame, "mobile_base_body_link", ovr2matrix(ts.HandPoses[ovrHand_Left].ThePose.Position,   lRot));
        tfPublisher->setTransform(right_frame, "mobile_base_body_link", ovr2matrix(ts.HandPoses[ovrHand_Right].ThePose.Position, rRot));
    }
```
**Notice** only the else branch is shown because the variable `relative` is always equals to `false` (i.e. it is set in the constractor of the class and it does not change). 

What I want to propose is:
1. add a new input argument (e.g. `"tf_head_frame"`)
2. publish the head homogeneous transformation (the name of the frame will be the one specified into the configuration file)
3. use the parameter specified into `"tf_root_frame"` as the name of the base link instead of the hardcoded `"mobile_based_link"`.

What do you think? If you want I can open a PR

GiulioRomualdi, 2018-12-20:
cc @andreadelprete

DanielePucci, 2019-01-06:
CC @aerydna

aerydna, 2019-05-07:
this is potentially fixed by #1958