robotology/yarp #1952: yarp::os::RFModule could call close() after an unsuccessful configure()

PeterBowman, 2019-01-19:
`yarp::dev::PolyDriver`'s destructor calls `yarp::dev::DeviceDriver::close`, which means that any failure in the `yarp::dev::DeviceDriver::open` stage would result in a controlled cleanup (i.e. close ports, release memory, etc., as long as `close()` is told to do so). I'd expect `yarp::os::RFModule` to behave similarly; instead, now I need to call `yarp::os::RFModule::close` manually, either within my custom implementation of `yarp::os::RFModule::configure` (and prior to any `return false;` instruction) or as a result of checking the return value of `configure()` (which means that I cannot use the handy helper signature `runModule(yarp::os::ResourceFinder &rf)`).

Sample *main.cpp* code for the latter case, desired behavior:

```c++
yarp::os::RFModule mod; // actually instantiate a custom derived class
return mod.runModule(rf); // calls configure(rf) and runModule() if successful
```

`runModule(rf)` could take care of the outcome of `configure(rf)` and call `close()` automatically (or let the destructor handle this). Current workaround:

```c++
yarp::os::RFModule mod; // actually instantiate a custom derived class

if (mod.configure(rf))
    return mod.runModule();
else
    return mod.close() ? 0 : 1;
```

Note that `close()` is indeed called before the module closes, as stated in the documentation. However, it is not whenever `configure()` returns `false` (in which case the module is never run, but some ports could remain open and/or dynamic memory not be deallocated).

pattacini, 2019-01-20:
Hi @PeterBowman 

I think you're giving for granted the correspondence between `yarp::dev::PolyDriver` and `yarp::os::RFModule` behaviors, which instead are two different objects and thus can behave differently:
- The user does not see what `PolyDriver` does internally and has no access to `PolyDriver::close()`.
- The user has to fill in the methods of `RFModule` and as such he/she is free to call them wherever wanted, in particular for what regards the allocation/deallocation of resources. Furher, the `RFModule::close()` is called when the sequence `CTRL-C` is detected, not really when the module exits, an event that might happen also when there's a failure and hence no `CTRL-C` is provided.

Your request would enforce a new flow in the schedule of `RModule` activities that is not foreseen by the present user code and thus potentially may cause even crashes. Look below:

```c++
bool configure(...) override {
    Object *obj=new Object;
    i=0;
    if (!ok) {
        close();           // why not to use it here?
        return false;
    }
    return true;
}

bool close() override {
    delete obj;
    i++;
    return true;
}
```

In case of failure, `close()` would be called twice - one by the user and one by YARP itself - with the following two unexpected results:
- the second call to `delete obj` will cause a seg-fault upon closure and without any failure because the lazy user didn't put the guard `obj=nullptr` just afterward, assuming that it wasn't necessary.
- the variable `i` would assume the value `2` instead of `1`, which is fully meaningless in this case but accounts for any kind of operations the user meant to perform inside the `close()` and that now get executed twice.

Nicogene, 2019-01-22:
Hi @PeterBowman ,

time ago @apaikan and me dig into the `RFModule` class and this is the state machine we discovered behind its behaviour:
[image]

So If the `configure()` return false, `close()` should be called.

pattacini, 2019-01-22:
Hi @Nicogene 
This's just come surprising to me since I actually thought the opposite for a decade or so 😉  

The following snippet confirms `RFModule::close()` doesn't get called instead:
```c++

#include <yarp/os/Network.h>
#include <yarp/os/ResourceFinder.h>
#include <yarp/os/RFModule.h>
#include <yarp/os/LogStream.h>

class TestClose : public yarp::os::RFModule
{
    bool configure(yarp::os::ResourceFinder &rf) override
    {
        yDebug()<<"configuring";
        return false;
    }
    bool updateModule() override
    {
        yDebug()<<"running";
        return true;
    }
    bool close() override
    {
        yDebug()<<"closing";
        return true;
    }
};

int main(int argc, char *argv[])
{
    yarp::os::Network yarp;
    yarp::os::ResourceFinder rf;
    rf.configure(argc,argv);
    TestClose mod;

    yDebug()<<"starting";
    return mod.runModule(rf);
}
```

The output is in fact:
```
[DEBUG]starting 
[DEBUG]configuring 
[INFO]RFModule failed to open.
```

pattacini, 2019-01-22:
I got now the difference!
In `main()` you can do
either:
```c++
configure(rf);    // close() might called here?
runmodule();
``` 
or:
```c++
runmodule(rf);    // configure() is called inside but close() is not executed in case of failure
```

pattacini, 2019-01-22:
Uhm 🤔 I've modified the snippet above to cover the other case:
```c++
#include <cstdlib>
#include <yarp/os/Network.h>
#include <yarp/os/ResourceFinder.h>
#include <yarp/os/RFModule.h>
#include <yarp/os/LogStream.h>

class TestClose : public yarp::os::RFModule {
public:
    bool configure(yarp::os::ResourceFinder &rf) override {
        yDebug()<<"configuring";
        return false;
    }
    bool updateModule() override {
        yDebug()<<"running";
        return true;
    }
    bool close() override {
        yDebug()<<"closing";
        return true;
    }
};

int main(int argc, char *argv[]) {
    yarp::os::Network yarp;
    yarp::os::ResourceFinder rf;
    rf.configure(argc,argv);
    TestClose test;
    yDebug()<<"starting";
    if (test.configure(rf)) {
        return test.runModule();
    }
    else {
        return EXIT_FAILURE;
    }
}
```
whose output is:
```
[DEBUG]starting 
[DEBUG]configuring 
```

lornat75, 2019-01-22:
looking at the code close() is called inside runModule(), therefore it should *not* be called if you do:

```c++
if (configure(rf))   
   runmodule();
``` 
or simply:

```c++
runmodule(rf);
``` 

It is called if you explicitly call runModule() as in:
```c++
configure(rf))   
runmodule();
```
I guess this falls in the realm of undefined behavior, becuase runmodule is not intended to be called if configure() fails.

lornat75, 2019-01-22:
> Note that `close()` is indeed called before the module closes, as stated in the documentation. However, it is not whenever `configure()` returns `false` (in which case the module is never run, but some ports could remain open and/or dynamic memory not be deallocated).

I think the idea is that configure() itself has to take care of cleanup, in case of failure. 

The rationale being that the module calls close() only for a module that has been successfully configured (opened).

Nicogene, 2019-01-22:
> I guess this falls in the realm of undefined behavior, because runModule is not intended to be called if configure() fails.

I'm afraid that in that test we did we fell in this case :sweat_smile:
But it was almost 3 years ago and I don't remember how we tested it :disappointed:

PeterBowman, 2020-01-01:
I'll adhere to the following policy: if there exists a `close` method that needs to release unmanaged resources (dynamically allocated memory) or terminate stuff in an ordered manner (if using PolyDriver class members, e.g. close device A before device B), always define a class destructor that calls `close`, be it a `DeviceDriver` or an `RFModule` derived class. Also, make sure nothing bad happens if this `close` method is called several times (i.e. set dangling pointers to `nullptr`). Why is that:

* `PolyDriver::open` may fail to initialize a subdevice, but it does **not** call the subdevice's `close` method; instead, it is immediately destructed via `delete` ([ref](https://github.com/robotology/yarp/blob/b3dff81c3739112b8f65cfd808f129bcbf4e7aa5/src/libYARP_dev/src/yarp/dev/PolyDriver.cpp#L308)).

* If `RFModule::configure` returns `false`, as explained in the above comments, `close` will never be called, hence I also want to use a destructor here.

* `PolyDriver::close` will never close a wrapped device twice, but callers of `RFModule` can do that inadvertently because of the previous point: once after a successful `RFModule::configure` and a CTRL+C signal (it just stops execution flow and calls `close` before leaving `runModule`), and one more time on class destruction.

* Despite those two mechanisms being relatively different, this policy ensures I treat both `RFModule` and `DeviceDriver` constructs in a similar manner: `close()` on destruction, avoid dangling pointers.

closed 2020-01-17 by drdanz