robotology/yarp #1953: String nightmare in ConnectionReader/Writer

randaz81, 2019-01-21:
The following methods are problematic:
* yarp::os::ConnectionReader::expectText(const char terminatingChar = '\n')
* yarp::os::ConnectionWriter::appendRawString(const char* str,const char terminatingChar = '\n' )
* yarp::os::ConnectionWriter::appendString(const std::string& s)

Currently, the only proved way to receive a string is as: size+blob of bytes, e.g.
```
    int string_size = connection.expectInt32();
    this->map_id.resize(string_size);
    connection.expectBlock(const_cast<char*>(this->map_id.data()), string_size);
```

drdanz, 2019-01-23:
After some investigation, I agree that this part is a nightmare. I'm planning to:

* Replace `ConnectionWriter::appendString` with `appendText` (`appendString` will be deprecated)
* Replace `ConnectionWriter::appendRawString` with `appendString` (different signature) (`appendRawString` deprecated)
* Add `ConnectionReader::expectString`
* Cleanup string serialization all around in YARP code. (This includes changing code generated by thift, but since I have a **huge** refactor on the way, I'm going to change this later).

The goal is to have either `appendText` -> `expectText` or `appendString` -> `expectString`

Since there are pure interfaces involved, this is going to be a breaking change, but I searched these methods using github search, and they seem to be used only inside YARP.

Nicogene, 2019-02-26:
Fixed by #1972

PR #1972 (2019-02-26): Synchronize "Append" and "Expect" methods for strings
* `ConnectionWriter::appendString()` is deprecated in favour of `appendText()`.
* `ConnectionWriter::appendRawString()` is deprecated in favour of  `appendString()`.
* Added `ConnectionReader::expectString()`.

Fixes #1953
closed 2019-02-26 by Nicogene