robotology/yarp #1955: Problem with  GazeboYarpControlBoardDriver::getPidOutput()

triccyx, 2019-01-31:
**Describe the bug**
When using GazeboYarpControlBoardDriver::getPidOutput() it doesn't return the
expected value. Using instead GazeboYarpControlBoardDriver::getPidOutputs() resolves
the problem.

**To Reproduce**
 
  ```
 yarp::dev::IPidControl *ipid=nullptr;
 Property options;
 options.put("device", "remote_controlboard");
 options.put("local","/pidoutputlogger");
 options.put("remote", "/MYROBOTNAME/all_joints");   
 yarp::dev::PolyDriver drive;
 drive.open(options);
 bool ok=drive.view(ipid);

 const int myJointNumber=2;
 double ref,ref1;
 double all[50];
 ipid->getPidOutput(VOCAB_PIDTYPE_POSITION ,myJointNumber,&ref);
 ipid->getPidOutputs(VOCAB_PIDTYPE_POSITION ,all);
 ref1=all[myJointNumber];
 std::cout<<"Single:"<<ref<<" Multiple:"<<ref1<<std::endl;
```

**Expected behavior**
Both APIs should give the same results.
Instead we have:

Single:0 Multiple:0.06893
Single:0 Multiple:0.0617215
Single:0 Multiple:0.0590815
Single:0 Multiple:0.0565569
Single:0 Multiple:0.0548603
Single:0 Multiple:0.0561772

**Configuration:**
 - OS: Ubuntu 16.04
 - yarp version: YARP version 3.1.100+20181004.5+git6fc8daf
-  compiler: gcc 5.4.0

traversaro, 2019-01-31:
If you try the same code on a port of the robot different from the `all_joints` one, you are experincing the same issue?

triccyx, 2019-01-31:
I have switch to:

`options.put("remote", "/MYROBOTNAME/head");`

Single:0 Multiple:0.159547
Single:0 Multiple:-0.136432
Single:0 Multiple:0.127655
Single:0 Multiple:0.0755312
Single:0 Multiple:0.0826702
Single:0 Multiple:0.109589
Single:0 Multiple:0.13336
Single:0 Multiple:0.13164
Single:0 Multiple:0.119059

The result is the same.

traversaro, 2019-01-31:
To check if the issue is indeed at the YARP level, the most straightforward strategy probably is to just modify an existing test to check also this. 
As the `fakeMotionControl` device (https://github.com/robotology/yarp/blob/b7983c941952a3aafa7de7dd4aeeb574c1865799/src/devices/fakeMotionControl/fakeMotionControl.cpp#L1812) already implements the `IPicControl` interface, it should be sufficient to add a test on the `getPidOutputs` or `getPidOutput` methods in  `ControlBoardRemapperTest` : https://github.com/robotology/yarp/blob/ff713f369ff3a00aafd64d8105597e203f58b65a/tests/libYARP_dev/ControlBoardRemapperTest.cpp#L72 .

randaz81, 2026-11-06:
This problem is related to the gazebo controller implementation, since all tests in YARP, based on the fakeMotionControl device, are successful.

closed 2026-11-06 by randaz81