robotology/yarp #1961: On the behaviour of Bottle::get()

claudiofantacci, 2019-02-07:
I just came across with @xEnVrE to a subtle bug in my code involving the `Bottle::get()` method that I would like to discuss here. The `Bottle::get()` command has an out-of-bound prevention mechanism for which if you ask for a value that is out-of-bound you get a "null" value. For example, if you are getting an out-of-bound `double`/`float`/`int` value you get `0.0`/`0.0`/`0` 🤔.
[Here](http://yarp.it/classyarp_1_1os_1_1Bottle.html#af93427131b0860ec0a2c3cfe952f3f86) is the documentation of the method and [here](https://github.com/robotology/yarp/blob/81ca0b0458d66bdb1f48edf3f1267a136cfe8ca1/src/libYARP_OS/src/BottleImpl.cpp#L640-L643) is the implementation.

_As a personal opinion_, I find this implementation wrong and misleading.
If, for any reason, your code uses `Bottle` to read or manipulate data, for example encoder values, and your code has a bug for which you are reading out-of-bound data (and be honest, can easily happen), you really don't want your code to silently run hiding a potential dangerous problem, and even worse getting zeros that may be correct values in some scenarios and wrong in others. What you would probably want is a runtime error or an [exception](https://en.cppreference.com/w/cpp/error/out_of_range).

Note that (as reported in the documentation) what you should with the current implementation is to get the refernece `Value` returned by the `Bottle::get()` method, check it with [Value::isNull()](http://yarp.it/classyarp_1_1os_1_1Value.html#ac232fce34f45ca6b07d6b89d08bb2dab) and eventually use it. This is a practice that goes in the oppsoite direction of what, for example, the STL does with the [`at()`](https://en.cppreference.com/mwiki/index.php?search=at) methods and is somewhat similar to exception handling, but without exception. For this last observation, note that modern compilers handles exception with new models that have zero-cost, unless the exception is thrown (which should not happen and when it does it is most likely that your application is about to fail and an error/bug need to be corrected). A discussion about this can be found [here](https://stackoverflow.com/questions/13835817/are-exceptions-in-c-really-slow).

I'm open to any observation or discussion 😄 I just wanted to share with you this though.

randaz81, 2019-02-07:
I faced this kind of problem many times in my interfaces. 
When an invalid numeric type is involved, I personally like to generate a quiet not-a-number `e.g. std::nan("")` instead of using zeros. NaNs propagate very well along your code, making it very easy to identify the source of the issue.

drdanz, 2019-02-09:
Just for reference, `nan` works only with floating point values, it doesn't work for integers. Anyway, i think that the point here is that `get()` should not return from an invalid index...

claudiofantacci, 2019-02-09:
Thanks for your point of view. It seems that this issue got some attention and we might want to rething it's behaviour (and maybe extend the possible new behaviour to other YARP classes).