robotology/yarp #1965: Make yarp::sig::Vector movable

valegagge, 2019-02-13:
I think the Vector class should be movable, that is modernize it by writing move constructor and a move assignment operator.

barbalberto, 2019-02-27:
I'd suggest to implement also iterators and `begin()` `end()` functionalities, so that we could use `yarp vectors` within std algorithms.

Nicogene, 2019-02-27:
Already done here: #1908 😎

valegagge, 2019-03-07:
Hi @Nicogene,

if i'm not wrong the movable costructor and operator= are not implemented. 
I think there is a good reason for this. Could you explain to me?

thanks!!

Nicogene, 2019-03-13:
Sorry I badly explained myself, #1908 implements `begin()` and `end()`.
#1974 instead implements the move constructor and operator. Unfortunately there are some memory leaks that I didn't find out how to fix it.

PR #1974 (2020-01-23): Make MangedBytes and VectorOf movable
This PR makes `ManagedBytes` and `VectorOf` movable.

It fixes #1965.

Please review the code.
closed 2020-01-23 by drdanz