robotology/yarp #1966: Wrap IPWMControl and ICurrentControl in SWIG bindings

traversaro, 2019-02-15:
**Is your feature request related to a problem? Please describe.**
The YARP interfaces for motion control need to be adapted to be used in scripting languages via SWIG, see for example how the [`IPosiitonControl`](https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/bindings/yarp.i#L811) or [`IMotorEncoders`](https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/bindings/yarp.i#L966) interfaces are extended.

 The `IPWMControl` and `ICurrentControl` are currently wrapped in SWIG (https://github.com/robotology/yarp/blob/24eb21146091e4cc3d5bba0856f1ab8fe9b202df/bindings/yarp.i#L762) but they are not properly extended to expose all those methods. 

**Describe the solution you'd like**
An implementation that expose the `getNumberOfMotors` and other methods that are not automatically wrapped by SWIG in a usable form, similar to what done for other interfaces.

PeterBowman, 2020-07-03:
@traversaro I'm aiming to submit a patch (hopefully before the 3.4 release) targeting this issue. However, I noticed an inconsistency in the pointer-like signature of several methods.

For instance, this is how `bool yarp::dev::IPositionControl::getRefSpeed(int j, double * v)` is extended in order to properly translate output pointer data in languages that don't support such idiom:

https://github.com/robotology/yarp/blob/f91ceb574b133b0cd45981b959d536dd62ac2e5e/bindings/yarp.i#L814-L816

Here, a one-dimensional vector is expected as `data`. When working with Python bindings, we'd pass a `yarp.DVector(1)`.

In contrast, this is `bool yarp::dev::IEncoders::getEncoder(int j, double * v)`:

https://github.com/robotology/yarp/blob/f91ceb574b133b0cd45981b959d536dd62ac2e5e/bindings/yarp.i#L915-L920

So, two approaches have been followed to tackle this problem, respectively:

* Preserve boolean return status, accept a one-dimensional list or array and populate it with the expected output value.
* Ignore the boolean status flag, reduce the number of arguments so that the desired value is returned instead, signalize failure with a dummy return value (in this case, `0`).

Judging by the adoption rate in yarp.i, I guess the former is preferred. Would you agree and expect that methods from missing interfaces (in the scope of this issue and beyond) follow this approach? Are there any exceptions to this "rule", or is there any unwritten convention (I'd say the `getAxes`-like methods don't really need the boolean status and might as well be kept simple, i.e. return the desired value)? Regarding existing methods that embrace the latter solution, should we add the two-argument counterpart for consistency (and perhaps deprecate the old signature)?

drdanz, 2020-07-06:
@PeterBowman we are aiming at releasing YARP 3.4 at the end of the month, which means we are now in a "soft freeze" state, I think we can make an exception for bindings, but if you want this in 3.4, please submit the patch as soon as possible...

drdanz, 2020-07-15:
Fixed in master. Thanks @PeterBowman!

closed 2020-07-15 by drdanz