robotology/yarp #1973: matlab binding problem "no yarp_java in java.library.path"

AL458, 2019-02-26:
Hi all,

I compiled Java bindings following the instructions in 
https://www.yarp.it/yarp_swig.html#yarp_swig_compile

Afterwards, I updated both the classpath.txt and the librarypath.txt Matlab files by adding the class paths:

<install_dir>/share/yarp/java/yarp.jar
<install_dir>/share/yarp/java/yarp_matlab_java.jar

and the library path

<install_dir>/lib/jni

respectively in classpath.txt and librarypath.txt files.

When I try to load yarp java libraries from matlab:

>> yarp.matlab.LoadYarp;
Java exception occurred:
java.lang.UnsatisfiedLinkError: no yarp_java in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at yarp.matlab.LoadYarp.<init>(LoadYarp.java:23)

OS: Ubuntu 18.04
YARP version 3.1.0~dev
Matlab 2018a

Can anyone help?

vtikha, 2019-02-26:
Hi @AL458 
are you using the absolute path to the libraries and jar in the classpath.txt and librarypath.txt files?
Please amend the files to include the full path and restart matlab.

AL458, 2019-02-26:
Hi @vtikha,

yes, I'm already using absolute paths:

/usr/share/yarp/bindings/build/share/yarp/java/yarp.jar
/usr/share/yarp/bindings/build/share/yarp/java/yarp_matlab_java.jar

and 
/usr/share/yarp/bindings/build/lib/jni

thanks for answering

AL458, 2019-02-26:
Solved.

My fault, I erroneously added the "/usr/share/yarp/bindings/build/lib/jni" path also to the classpath.txt file. Once removed, everything worked.

Thanks!

closed 2019-02-26 by AL458

vtikha, 2019-02-26:
Excellent