robotology/yarp #1979: fromCvMat in camcalib produce segFault

gonzalezJohnas, 2019-03-18:
This bug has been found and solve   with the help of  Andrea Ruzzenenti (@aerydna) 

**Describe the bug**
When running camCalib from icub-main and more specifically  in the "PinholeCalibTool::apply" function produce a null image after "fromCvMat<PixelRgb>" resulting in a segfault.

We found that the issue is in reality in  libYARP_sig/src/Image.cpp for the function Image& Image::operator=(Image&& other).

**To Reproduce**
Connect camCalib module to yardev (grabberDual)

**Expected behavior**
The image returned from fromCvMat<PixelRgb> after applying the calibration with remap is not valid so accessing is pixel produce a segfault.

**Configuration :**
 - OS: Ubuntu 18.04
 - yarp version: 3.0 branch devel
-  compiler: gcc, g++ version 7.3.0

**Temporary fix**
The actual bug happen in the libYARP_sig/src/Image.cpp : 
Image& Image::operator=(Image&& other) 
    Image moved(std::move(other));
    std::swap(moved.implementation, implementation);
    synchronize();
    return *this;

When performing the swap if *moved.implementation* is not set to nullptr, the Image destructor is called on *moved*  destroying also the implementation of the current object (this).

A way to fix it is to explicitly set moved.implementation to nullptr before the synchronize or before the swap.

For more details you can contact me or Andrea Ruzzenenti (@aerydna)

drdanz, 2019-03-19:
Thanks for reporting this.
It's fixed in `devel` now.

closed 2019-03-19 by drdanz

Nicogene, 2019-03-29:
The move operators were correct, the commit that add `nullptr` has been reverted since it introduced memory leak.

The bug was in `camCalib`, it has been fixed by https://github.com/robotology/icub-main/pull/572

PR #1980 (2019-03-19): YARP_sig: fix move operator of Image
It fixes #1979 

The best should be add a test to reproduce the bug fixed, but til now I was not able to reproduce it without opencv.

Please review code.