robotology/yarp #1984: rpLidar2 SDK fails to compile on travis (clang builds)

drdanz, 2019-03-22:
See for example:

https://travis-ci.org/robotology/yarp/builds/509448269

```
Scanning dependencies of target rpLidar2_sdk
[ 55%] Building CXX object src/devices/rpLidar2/CMakeFiles/rpLidar2_sdk.dir/__/__/__/extern/rplidar/sdk/src/rplidar_driver.cpp.o
/home/travis/build/robotology/yarp/extern/rplidar/sdk/src/rplidar_driver.cpp:1704:10: error: case value evaluates to -1, which cannot be narrowed to type 'unsigned long' [-Wc++11-narrowing]
    case rp::hal::Event::EVENT_TIMEOUT:
         ^
/home/travis/build/robotology/yarp/extern/rplidar/sdk/src/rplidar_driver.cpp:1733:10: error: case value evaluates to -1, which cannot be narrowed to type 'unsigned long' [-Wc++11-narrowing]
    case rp::hal::Event::EVENT_TIMEOUT:
         ^
2 errors generated.
make[2]: *** [src/devices/rpLidar2/CMakeFiles/rpLidar2_sdk.dir/__/__/__/extern/rplidar/sdk/src/rplidar_driver.cpp.o] Error 1
make[1]: *** [src/devices/rpLidar2/CMakeFiles/rpLidar2_sdk.dir/all] Error 2
make: *** [all] Error 2
```

A similar problem was fixed in #1792 but in this case the problem was in our code, not in the SDK.
Latest release ([1.10.0](https://github.com/Slamtec/rplidar_sdk/releases/tag/release%2Fv1.10.0)) does not seem to fix the issue.

drdanz, 2019-04-11:
This is actually an upstream bug. Reported upstream: Slamtec/rplidar_sdk#14

closed 2019-04-11 by drdanz

l376571926, 2022-07-25:
> See for example:
>
> https://travis-ci.org/robotology/yarp/builds/509448269
>
> ```
> Scanning dependencies of target rpLidar2_sdk
> [ 55%] Building CXX object [...]

case (unsigned long)rp::hal::Event::EVENT_TIMEOUT: